PX4/PX4-ECL #30: Magnetic declination always starts at zero

priseborough, 2016-01-30:
The EKF has to learn the declination starting from an initial value of zero for each flight. We should calculate an initial value using the WGS-84 position or remember the value the end of each flight.

LorenzMeier, 2016-01-30:
The API for that is in PX4. You just have to call it.

priseborough, 2016-01-31:
Capturing the requirements here:

1) The startup declination shall be taken from a user adjustable magnetic declination parameter.
2) When the WGS-84 NED origin is set, the EKF magnetic field states and heading shall be updated using the declination returned by get_mag_declination(lat, lon). A configuration parameter shall be provided to the user to enable this yaw and field update to be inhibited. Note:  this update will introduce a step change in heading angle when the EKF origin is set. This will cause a UX problem when taking off without GPS and gaining GPS snd setting the origin in flight due to the 'kick' in yaw angle. This will require publishing the yaw angle change so that the control loops can apply a set-point correction.
3) The user shall be provided with a configuration parameter option of setting and saving the magnetic declination parameter automatically:

Option 1: Do not update and save
Option 2: Update to the value returned by get_mag_declination(lat, lon) and save. (default).
Option 3: Update to the value learned by the EKF during up and away flight and save

priseborough, 2016-02-09:
A simplified version that does not offer option 3 is presented in PR #42

closed 2016-03-13 by priseborough