PX4/PX4-ECL #34: Incomplete magnetometer and heading fusion tuning parameter implementation

priseborough, 2016-01-31:
The heading fusion observation noise parameter EKF2_HEAD_NOISE adjusts the observation variance, not the observation noise. It should be adjusting noise and have units of rad
The magnetometer noise parameter EKF2_MAG_NOISE ias actually adjusting the magnetic field state process noise, nor the magnetometer observation noise. It should be renamed.
There is no parameter to adjust the magnetometer observation noise
There is no parameter to control which magnetometer fusion method is used. A suggestion is to provide three options
1) Auto selection (default)
2) Always used heading fusion
3) Always use 3-axis field fusion

priseborough, 2016-02-03:
Addressed by #36

tumbili, 2016-02-06:
Applied to master, closing.

closed 2016-02-06 by tumbili