PX4/PX4-ECL #44: Estimate baro offset

LorenzMeier, 2016-02-11:
@priseborough This is to track the baro offset estimation.

CarlOlsson, 2016-09-19:
[Here](http://logs.uaventure.com/view/npxmVkZpUTnGTGgEYFaQpU) is a log from a 4 minute flight which shows the baro drifting 1.4 meters.  What is also interesting is that the global position estimate drifted 2.4 meters, i.e. more than the sensor itself.

tumbili, 2016-09-23:
@CarlOlsson  Even here, baro and the estimated height align really well and the innovations look normal.
I aligned the estimated height with the baro at the start of the log and swapped one sign to compare them.
[image]

CarlOlsson, 2016-09-23:
@tumbili yes you are right that they align pretty good, but the estimate is still drifting more than the baro (due to accelerometer noise?), so if this is normal behavior maybe we could just decrease the baro noise to trust it more?

priseborough, 2016-10-26:
@CarlOlsson If you do that then varying baro positional errors will cause larger changes in estimated height during manoeuvres or during transition in and out of ground effect. This is a compromise tune that needs to work across a wide range of setups.

bresch, 2021-08-17:
Addressed by https://github.com/PX4/PX4-Autopilot/pull/17944

closed 2021-08-17 by bresch