PX4/PX4-ECL #77: TODO list for airspeed fusion

CarlOlsson, 2016-03-14:
- [ ] Increase the airspeed readout rate to 14 Hz and test in SITL
- [ ] Optimize the calculation of the covariance matrix by taking advantage of the sparseness in H
- [ ] Add protection against the denominator in line 76 being too small
- [ ] Move the check for if we want to fuse airspeed somewhere higher up
- [ ] Implement the logic of when to fuse airspeed, i.e. not on the ground etc.
- [ ] Make sure the wind states and covariances are initialized correctly when we start fusing airspeed

LorenzMeier, 2016-04-26:
@CarlOlsson Where are we standing with this?

CarlOlsson, 2016-04-28:
I think all of the remainging points except the calculation of the covariance matrix were resolved by @tumbili ?

tumbili, 2016-04-28:
@CarlOlsson Yes, we can close this. Everything is in now, we just have to go on with testing.

closed 2016-04-28 by tumbili