PX4/PX4-ECL #91: EKF2 vertical height diverged

tumbili, 2016-04-11:
Happened on a test flight with wing wing and Pixracer

http://logs.uaventure.com/view/EDXp7BmVQsSsboydZMKVWW

First insights from discussion with @priseborough:

 - [ ] when local position state is reset the corresponding velocity states needs to be reset as well (not the case now)
  - [ ] the high innovations and the resulting divergence of the vertical height could have been caused by timing issues (need to check the performance results)

priseborough, 2016-04-11:
https://github.com/PX4/ecl/pull/92

priseborough, 2016-04-11:
I have tested its ability to recover from temporary large errors on IMU z accel data. However we need to give more thought to the desired system behaviour during a height sensor timeout and immediately following a reset.

priseborough, 2016-04-20:
I have incorporated furthers changes to height reset behaviour here and am i the process of testing it:

https://github.com/priseborough/ecl/tree/pr-ekf2HeightReset

We need a means of inducing sensor faults in SITL

closed 2016-05-03 by priseborough