PX4/PX4-ECL #100: Flag for height measurement update when using external vision

devbharat, 2016-04-19:
When using external vision, external vision height is never used for height update as the baro flag does not go "off" and the if(baro) else if(external vision) fails to use external vision height. Need to reset baro flag when using external vision correct. 

For now, changing 

    if (_fuse_height) {
        if (_control_status.flags.baro_hgt) {
            fuse_map[5] = true;
            // vertical position innovation - baro measurement has opposite sign to earth z axis
            _vel_pos_innov[5] = _state.pos(2) + _baro_sample_delayed.hgt - _baro_hgt_offset - _hgt_sensor_offset;
            // observation variance - user parameter defined
            R[5] = fmaxf(_params.baro_noise, 0.01f);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.baro_innov_gate, 1.0f);

        } else if (_control_status.flags.gps_hgt) {
            fuse_map[5] = true;
            // vertical position innovation - gps measurement has opposite sign to earth z axis
            _vel_pos_innov[5] = _state.pos(2) + _gps_sample_delayed.hgt - _gps_alt_ref - _hgt_sensor_offset;
            // observation variance - receiver defined and parameter limited
            // use scaled horizontal position accuracy assuming typical ratio of VDOP/HDOP
            float lower_limit = fmaxf(_params.gps_pos_noise, 0.01f);
            float upper_limit = fmaxf(_params.pos_noaid_noise, lower_limit);
            R[5] = 1.5f * math::constrain(_gps_sample_delayed.vacc, lower_limit, upper_limit);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.baro_innov_gate, 1.0f);

        } else if (_control_status.flags.rng_hgt && (_R_to_earth(2, 2) > 0.7071f)) {
            fuse_map[5] = true;
            // use range finder with tilt correction
            _vel_pos_innov[5] = _state.pos(2) - (-math::max(_range_sample_delayed.rng * _R_to_earth(2, 2),
                                 _params.rng_gnd_clearance));
            // observation variance - user parameter defined
            R[5] = fmaxf(_params.range_noise, 0.01f);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.range_innov_gate, 1.0f);
        }

    }

to

    if (_fuse_height) {
        if (_control_status.flags.baro_hgt && !_control_status.flags.gps && !_control_status.flags.ev_pos && !(_control_status.flags.rng_hgt && (_R_to_earth(2, 2) > 0.7071f))) {
            fuse_map[5] = true;
            // vertical position innovation - baro measurement has opposite sign to earth z axis
            _vel_pos_innov[5] = _state.pos(2) + _baro_sample_delayed.hgt - _baro_hgt_offset - _hgt_sensor_offset;
            // observation variance - user parameter defined
            R[5] = fmaxf(_params.baro_noise, 0.01f);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.baro_innov_gate, 1.0f);

        } else if (_control_status.flags.gps_hgt) {
            fuse_map[5] = true;
            // vertical position innovation - gps measurement has opposite sign to earth z axis
            _vel_pos_innov[5] = _state.pos(2) + _gps_sample_delayed.hgt - _gps_alt_ref - _hgt_sensor_offset;
            // observation variance - receiver defined and parameter limited
            // use scaled horizontal position accuracy assuming typical ratio of VDOP/HDOP
            float lower_limit = fmaxf(_params.gps_pos_noise, 0.01f);
            float upper_limit = fmaxf(_params.pos_noaid_noise, lower_limit);
            R[5] = 1.5f * math::constrain(_gps_sample_delayed.vacc, lower_limit, upper_limit);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.baro_innov_gate, 1.0f);

        } else if (_control_status.flags.rng_hgt && (_R_to_earth(2, 2) > 0.7071f)) {
            fuse_map[5] = true;
            // use range finder with tilt correction
            _vel_pos_innov[5] = _state.pos(2) - (-math::max(_range_sample_delayed.rng * _R_to_earth(2, 2),
                                 _params.rng_gnd_clearance));
            // observation variance - user parameter defined
            R[5] = fmaxf(_params.range_noise, 0.01f);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.range_innov_gate, 1.0f);
        } else if (_control_status.flags.ev_pos) {
            // warnx("fusing ext_visn_hight");
            fuse_map[5] = true;
            // calculate the innovation assuming the external vision observaton is in local NED frame
            _vel_pos_innov[5] = _state.pos(2) - _ev_sample_delayed.posNED(2);
            // observation variance - defined externally
            R[5] = fmaxf(_ev_sample_delayed.posErr, 0.01f);
            R[5] = R[5] * R[5];
            // innovation gate size
            gate_size[5] = fmaxf(_params.baro_innov_gate, 1.0f);
        }

    }

enables use of external vision height.

LorenzMeier, 2016-04-20:
@priseborough Could we get this in? We're pushing hard on the vision integration and I don't want to loose momentum or create branch drift.

devbharat, 2016-04-20:
https://github.com/PX4/ecl/pull/101

priseborough, 2016-05-03:
@devbharat and @tumbili 

Pulling in external vision PR #106 is blocked until we have the ability to log and replay external vision data as that is the only way I can test it.

devbharat, 2016-05-03:
I have made the PR for that. I'll send the log to Roman and once that's in, we could get this in too.

closed 2016-05-17 by priseborough