PX4/PX4-ECL #103: local_position reference alt when using External vision

devbharat, 2016-04-21:
When the GPS is connected to the pixhawk and the GPS module is started, but external vision is enabled for position and altitude instead of GPS, the reference altitude of local position somehow also drifts with the GPS value(indoors). Turning off the GPS module or disconnecting it solves the issue.

I suspect its the estimators "_gps_alt_ref" value is leaking in from GPS somehow.

priseborough, 2016-05-17:
Can you please provide a replay log from a flight where this behaviour is demonstrated.

devbharat, 2016-05-17:
This was from the code long ago, I haven't been up to testing recently. I'll see if I can recreate it and post update here.

devbharat, 2016-05-23:
I really do not have the resources now to conduct this test outdoors. Either we let this hang till I get the dataset with the temperory solution being disabling GPS when using vision, or we close it and reopen if the issue is observed again.

closed 2016-05-23 by priseborough