PX4/PX4-ECL #109: External vision XY initialization and tracking

devbharat, 2016-04-26:
At initialization LPOS.X and LPOS.Y take way too long to converge to values sent by external vision. (Probably because aircraft is stationary by not at origin when EKF2 starts, so external vision is sending in non zero values.)

[image]

And they oscillate on external vision reset to new value:
[image]

External vision data is fused with 1cm variance.

devbharat, 2016-04-26:
Full log file can be found at http://wikisend.com/download/174296/log001.px4log

priseborough, 2016-05-17:
Fixed in master

closed 2016-05-17 by priseborough