PX4/PX4-ECL #111: Snapdragon: Poor altitude estimation with range sensor

ndepal, 2016-04-27:
I am trying to get the snapdragon flying in altitude control based on altitude estimation with the TeraRanger One. However, `local_position_NED.z` is very inaccurate and seems to depend a lot on the quality of the acceleration measurements. 

I'm testing this the following way: I've hardcoded the TeraRanger driver to output a distance of 0.3m. `EKF2_RNG_NOISE` is set to 0.0001. The quad sits stationary on the ground. After a few seconds, I arm, after a few more seconds I increase the throttle.

While disarmed, `local_position_NED.z` already oscillates too much for my taste given the low measurement uncertainty. After arming, oscillation increases a lot (+/-10cm). See this log: http://logs.uaventure.com/view/5SfAq53SPzWRjmnxWUHF3U#Local_Z_PLOT

This oscillation appears largely independent of `EKF2_RNG_NOISE` and `EKF2_ACC_NOISE`. The problem seems to be the complementary filter. `EST0.s8` tracks the range measurements quite reasonably, it's just `LPOS.Z` that oscillates.
I've confirmed this by changing [this line](https://github.com/PX4/ecl/blob/master/EKF/ekf.cpp#L651) to:
```
_output_new.pos += _vel_corr * imu_new.delta_vel_dt;

```
Now the estimated altitude looks much better: http://logs.uaventure.com/view/635VkhfaZqYAmDcM5ZeuDX#Local_Z_PLOT

Thus, the problem is the noise in `delta_vel_NED(2)`, which cannot be tuned away by any EKF2 parameters, as far as I can see.

What is the suggested action to improve the altitude estimate?

LorenzMeier, 2016-04-27:
@priseborough This is for you

priseborough, 2016-04-27:
The first action is to add EV messages to the logging and replay test harness so I can diagnose and fix tuning and other issues. The EV PR https://github.com/PX4/ecl/pull/106 will not be pulled in until there is corresponding PR for the logging and replay in PX4/Firmware.

ndepal, 2016-04-28:
Here is a replay log of a flight with the snapdragon and tera ranger. In the second half, we were flying in altitude hold. You can see that `LPOS.Z` looks quite constant, while `EST0.s8` oscillates and more closely matches `RPL4.rng`. The constant `LPOS.Z` can be attributed to the altitude controller doing its job. However, the quad was not stable in height but rather did what  `EST0.s8` and `RPL4.rng` tell you.

http://logs.uaventure.com/view/JfRi8kAtn9D9Ezi6Z9uBaa

priseborough, 2016-04-30:
I've spent some time with the replay and have come to the conclusion that it is caused by a combination of factors.

1) Poor quality accelerometer data as evidenced by large range  finder innovations
2) Insufficient gain in the output complementary output filter
3) tuning on the EKF that does not allow for the amount of accelerometer error.

I have pushed some changes to ecl master which improve the tracking of the complementary filter. If you try these in combination with an increase in the value of EKF2_ACC_NOISE to 0.5, then you should get better results.

It is possible that high vibration levels combined with sensor scale factor errors and non-linearties are responsible for the degraded delta velocity accuracy . Some suggested activities are:

1) Carriage test without motors running to verify filter performance without vibration
2) Logging and inspection of 8kHz raw data to determine the noise spectra and peak levels.
3) Review of IMU processing algorithms used to down-sample and compensate for coning and sculling errors.

The parameters in http://logs.uaventure.com/view/JfRi8kAtn9D9Ezi6Z9uBaa show the following extreme parameter values were used:

EKF2_RNG_NOISE = 0.00999999977648
EKF2_RNG_GATE = 50.0

Can you confirm that these were only used for testing in replay and were not used for the flight?

ndepal, 2016-05-05:
Thanks for looking into this.

Regarding the hand held test you propose: We did this and the performance looks significantly better. However, what we do notice is an overshoot of the altitude estimation whenever we change the quad's altitude (i.e. overshoot in the step response). We observe the same on a Pixhawk.

Regarding the EKF2 parameters. I'm aware that these are extreme. I played around with these values in an attempt to improve the estimated altitude. While I did not see any real change in `LPOS.Z`, it does improve `EST.s8`. These parameters were set during flight as well.

I will run some tests with the changes you have pushed to ecl as soon as I can and report back.

ndepal, 2016-05-10:
I have tested the newest master on snapdragon today. I can gladly report that the altitude estimation looks significantly better now.

I've uploaded a log of today's flight: http://logs.uaventure.com/view/SCaV4ZGFbsFcVHdyHZnis5
Between seconds 90 and 160 I am in altitude hold.
The quad still oscillates up and down a good 20 cm. I would expect that performance could be better, given that I'm estimating altitude with a range sensor. However, I also have not done any tuning of the altitude controller.

If you feel that the performance of the altitude estimator is now satisfactory in this log, feel free to close this issue.

Thanks again for your help.

priseborough, 2016-05-10:
It's a vibration /flight control problem. Height sensor innnovation on the EKF are less than 5cm in flight, but noise levels in the control loops are very high. Better isolation of the snapdragon board or filtering of accel data used for control (not navigation) is required - what is your isolation solution? It looks like your isolator mount natural frequency in the roll degree of freedom is being excited by the motors/blade passage frequency and overall your isolators are too stiff.

Vertical EKF in-flight innovations low
[image]

Roll rate noise very high
[image]

Vertical accel noise levels high
[image]

Range finder, EKF vertical states state and output filter vertical states consistent
[image]

closed 2016-05-10 by priseborough