PX4/PX4-ECL #113: Standalone build doesn't work

julianoes, 2016-04-27:
I'm trying to build this following the instructions of the [README.md](https://github.com/PX4/ecl/blob/master/README.md).

```
~/src/Firmware/src/lib/ecl/build
[127] % make
[  8%] Building CXX object CMakeFiles/ecl.dir/estimator_interface.cpp.o
In file included from /home/julianoes/src/Firmware/src/lib/ecl/EKF/geo.h:49:0,
                 from /home/julianoes/src/Firmware/src/lib/ecl/EKF/estimator_interface.h:45,
                 from /home/julianoes/src/Firmware/src/lib/ecl/EKF/estimator_interface.cpp:46:
/home/julianoes/src/Firmware/src/lib/ecl/EKF/mathlib.h:44:23: fatal error: Eigen/Dense: No such file or directory
compilation terminated.
CMakeFiles/ecl.dir/build.make:62: recipe for target 'CMakeFiles/ecl.dir/estimator_interface.cpp.o' failed
make[2]: *** [CMakeFiles/ecl.dir/estimator_interface.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ecl.dir/all' failed
make[1]: *** [CMakeFiles/ecl.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

This is on Ubuntu 16.04.

julianoes, 2016-04-27:
Hang on, I might have to read the instructions more carefully.

julianoes, 2016-04-27:
No, the Eigen3 path is hard-coded for Mac. A potential fix is in #114.

mcsauder, 2016-04-27:
Hi, I had this trouble in another project after upgrading to 16.04 as well.  I had to add symbolic links in the following manner:

    cd /usr/include/
    sudo ln -sf eigen3/Eigen Eigen
    sudo ln -sf eigen3/unsupported unsupported

This fixed the issue for me in my other projects this week, (also, I had encountered this issue previously back in Ubuntu 13.04 and 13.10).  I hope this might help.

julianoes, 2016-04-28:
@mcsauder: thanks for the comment. I prefer not having to do symlinking and using cmake's FindEigen3 to look for it.

See how it's done in #114. Does that work for you?

mcsauder, 2016-04-28:
Hi @julianoes,

I agree with your sentiment on avoiding sym links, I only employ them as a last resort.  Unfortunately, I already have FIND_PACKAGE(Eigen3) in the top level CMakeLists.txt file for this and two other projects using Eigen that began failing compile after the upgrade to 16.04.  :/

As an additional detail that might be worth investigating, if Eigen is installed in the typical manner through apt-get it will install to the ~/usr/include/eigen3 directory, however if you clone Eigen from the source repo and execute a make install, (without altering the default install directory), it will install to your ~/usr/local/include/ directory.  On my machine, with Eigen installed to the ~/usr/local/include/eigen3/... directory, my projects will compile without the sym links, however they will not compile without the sym links when Eigen is installed only in the /usr/include/eigen3 directory, even with FIND_PACKAGE(Eigen3) is called out in the top level CMakelists.txt file and even if I have used apt-get to remove Eigen and re-installed it.

At the moment I am contemplating a fresh install of 16.04 to see if it behaves differently than the upgrade from 15.10.  I will let you know if I get that accomplished and what results after a clean install.  Hopefully a bit of this is some use to you as well.

-Mark

julianoes, 2016-04-28:
I think it is common practice to have the package from the distribution in /usr/include and install libraries compiled/downloaded yourself in /usr/local/include. I would also guess that FIND_PACKAGE(Eigen3) should check both paths.

It works fine for me on a fresh 16.04 with `libeigen3-dev` installed with apt-get.
It shows up like this:
```
locate eigen3/Eigen/Dense
/usr/include/eigen3/Eigen/Dense
```

I think you should be able to fix it without re-installing everything.

mcsauder, 2016-05-10:
Hi @julianoes,

I just wanted to report back that a new install of Ubuntu 16.04 resolved the issue of not being able to find eigen, so all of my machines are behaving correctly now without symbolic links.  

I have no idea what error existed on my machine, but all is well now.  Thanks for all of the follow-up and advice!

-Mark

julianoes, 2016-05-10:
@mcsauder: nice, thanks for the heads-up!

priseborough, 2016-05-17:
So where do we stand with this issue now?

julianoes, 2016-05-17:
Yep sorry fixed with #114.

closed 2016-05-17 by julianoes