PX4/PX4-ECL #121: altitude estimate diverges when fusing airspeed data

CarlOlsson, 2016-05-05:
From two flights with the wingwing
[image]
[image]

The raw airspeed sensor data doesn't look so nice so will try to replace the airspeed sensor and level shifter with a new one from AUAV and try again.

LorenzMeier, 2016-05-05:
@CarlOlsson Can you have a look at this here? The code before Daniel's fix looks wrong to me, and I would not be surprised if that results in what you're seeing here:
https://github.com/PX4/ecl/pull/120/files

LorenzMeier, 2016-05-05:
@CarlOlsson Can you share the raw log file via Log muncher?

And can you please do a flight again with replay logging?

CarlOlsson, 2016-05-05:
Will do the two flights tomorrow and share the logs. Just want to make sure the sensor is working correctly first

CarlOlsson, 2016-05-06:
First flight today
[image]
[image]
Logs here:
http://logs.uaventure.com/view/F2SnAfAwav576JyfqmLeQe
http://logs.uaventure.com/logmuncher/charts/F2SnAfAwav576JyfqmLeQe/logmuncher.log

CarlOlsson, 2016-05-06:
Made a second flight, deactivated airspeed fusion completely. Ekf still diverges.
[image]
http://logs.uaventure.com/view/eVXyzfzw2oryqRJNvuumC3
http://logs.uaventure.com/logmuncher/charts/eVXyzfzw2oryqRJNvuumC3/logmuncher.log

LorenzMeier, 2016-05-06:
Did you create the replay logs as requested? Without that more flights are relatively pointless.

CarlOlsson, 2016-05-10:
Yesterday I flight tested ecl master and this pr PX4/firmware#4484
[image]
[image]
Unfortunately the RPL logs seems corrupt will try to fix it as soon as possible

CarlOlsson, 2016-05-10:
Plots: http://logs.uaventure.com/view/2y6aLea2BuiXQSeXe9zYHZ
Log: http://logs.uaventure.com/logmuncher/charts/2y6aLea2BuiXQSeXe9zYHZ/logmuncher.log

CarlOlsson, 2016-05-10:
I remounted the pixracer such that it is damping vibrations.
Now the altitude is not diverging anymore
[image]

closed 2016-05-10 by CarlOlsson

CarlOlsson, 2016-05-10:
[image]
[image]
[image]