PX4/PX4-ECL #126: RPL logs with airspeed sensor

CarlOlsson, 2016-05-11:
Did two flights with the wingwing
[image]

**First flight**
Plots: http://logs.uaventure.com/view/8gD2HCcvqjobKm5Bq46Yei
Log: http://logs.uaventure.com/logmuncher/charts/8gD2HCcvqjobKm5Bq46Yei/logmuncher.log

**Second flight**
Plots: http://logs.uaventure.com/view/hbXUjHrcGHQm9Dpuxxcv5d
Log: http://logs.uaventure.com/logmuncher/charts/hbXUjHrcGHQm9Dpuxxcv5d/logmuncher.log

@priseborough @tumbili 
Would be very happy for any comments on the logs

priseborough, 2016-05-11:
What version of the code were you using and have you isolated your pixracer from vibration yet?

CarlOlsson, 2016-05-12:
Yes the board is isolated #121
I was using this branch
0d377dc5f4ddbfd05af5d6e66cdac38e0ff49a3d
https://github.com/CarlOlsson/Firmware/tree/add_airspeed_to_rpl
I have a problem with the replay logs. If I activate airspeed fusion I still don't get any airspeed innovations in the logs. Even though I get it with the same code in SITL. @tumbili thought it could be something with how the flags in the estimator are logged. My idea now is to do another test with current master and post an issue about it if it persists.

CarlOlsson, 2016-05-12:
See PX4/Firmware#4515

priseborough, 2016-05-12:
What is the highest linear or rotational natural frequency of your isolation system and is there anything that could be bypassing it?

CarlOlsson, 2016-05-12:
The pixracer is mounted on 4 silicone tubes like this:
[image]
I am not sure if that answers your question? Otherwise please explain a bit more and I will do some measurements if required

CarlOlsson, 2016-05-12:
ekf2 replay is finally working with airspeed.
[image]
@priseborough @tumbili I am a bit confused of the innovations before we are airborne, would appreciate any comments!

CarlOlsson, 2016-05-13:
Here is a mission flown in SITL. The vtol takes of flies for some time, transitions to hover and then transitions back to cruise.
[image]
The innovations of the fused tas shoud be zero during hover but they are not

tumbili, 2016-05-13:
@CarlOlsson Remember that when nothing is fused the innovation will just remain constant at the last value. Maybe this is why you are seeing innovations during hover.

CarlOlsson, 2016-05-13:
Yea you are right. No innovations are published during hover but the postprocessing sw fills up the missing data points with the last value. Would you recommend to keep it this way?

priseborough, 2016-05-17:
@CarlOlsson, those mounts are too stiff for a board of such low mass. You need to get the vibration modes below 50Hz. I use a thin ply mounting plate with 4 8x8x6 mm low density latex foam squares as shown below:
[image]

closed 2016-05-17 by priseborough