PX4/PX4-ECL #140: Robustness to onboard vibration

LorenzMeier, 2016-05-18:
@priseborough @ndepal @CarlOlsson @boosfelm This is to discuss the effects of medium to strong vibration we saw on EKF2 and the relationship to other estimators which seemed to survive these better or at least hide the problem in a way that we felt they were fine to operate.

Could you share recent replay logs of these and maybe also a baseline flight with EKF1?

CarlOlsson, 2016-05-18:
Today we flew the wingwing four times. Estimator diverged every flight, the pixracer is mounted with isolation but we already discussed that it could be to little. 

We also flew our system with ekf2 and airspeed fusion. I haven't had time to look at the data but I think it was all fine since we did tecs etc. will update here tomorrow with all the logs etc

priseborough, 2016-05-19:
To assist with the investigation I have created a branch where you can switch to using the rate and accel output from the IMU driver instead of the delta angle and velocity output and also select a different IMU.
https://github.com/PX4/Firmware/tree/ekf2ImuSampleInvestigation

ndepal, 2016-05-19:
I have the following logs for the snapdragon. In all of them I'm interested in `LPOS.Z` tracking the range measurements, which are set to be used for altitude estimation.

1) Hand held altitude steps 50 cm - 100 cm - 150 cm: http://logs.uaventure.com/view/SgLGKSantZPuGRGyaokEn9

2) Flight test: http://logs.uaventure.com/view/abRQ9KEnGNMY5Z2mBKarEP

The problems I have are drift even while the quad is sitting on the ground, disarmed as well as significant oscillations of `LPOS.Z` around the range data when armed/in flight. This can be seen in this plot of the log 2), where at first the quad is disarmed, then armed 90s

[image]

I will test the new branch with the EKF2_IMU_MODE options as well as other estimators next.

priseborough, 2016-05-19:
Here's the replay of LPOS, range and estimator state for the hand test
[image]

priseborough, 2016-05-19:
Here's clear evidence of the output observer tracking degrading as vibration levels increase:
[image]

[image]

ndepal, 2016-05-19:
I have made several logs with the https://github.com/PX4/Firmware/tree/ekf2ImuSampleInvestigation branch:

The sign of the distance measurements has been inverted in the below figures.

1) `EKF2_IMU_MODE 0`, hand held: http://logs.uaventure.com/view/A3Q2qHRHfRWMjEvVdZ9LNJ
[image]

2) `EKF2_IMU_MODE 1`, hand held: http://logs.uaventure.com/view/EN4kFiBz5ZybKujo4QH78U
Observation: Both `EST0.s9` and `LPOS.Z` react poorly to range data outliers (e.g. sec 81).
[image]

3) `EKF2_IMU_MODE 0`, manual flight: http://logs.uaventure.com/view/gNbFqLgVmUc73fWAWAszwG
[image]

4) `EKF2_IMU_MODE 1`, manual flight: http://logs.uaventure.com/view/5MUbBjRnhvkY6aK3YeX525
[image]

5) `EKF2_IMU_MODE 0`, alt ctl flight: http://logs.uaventure.com/view/ofn8G8zhNyyjJdaD3QvUN5
Observation: The altitude estimation diverges as I set down the quad to arm (sec 80). In the phase before that, where I am carrying the quad, the altitude estimates do not track the range sensor data. Fortunately the reset was successful and the estimate was reasonable again during altitude hold (I did not see this divergence as it happened as I fly without view of my computer).
[image]

This is now with the snapdragon mounted underneath the Bebop frame, suspended at three points (2 in the front on either side of the camera, one in the rear) with foam. With this, the gyro data looks as follows:
[image]

Seconds 10 - 30 are hand held, walking; seconds 42 - 45 are armed, idle; seconds 45 - 140 are in flight.

priseborough, 2016-05-19:
Significant reductions the time constants on the complimentary filter did reduce the output observer differences slightly, eg
EKF2_TAU_VEL 0.1
EKF2_TAU_POS 0.05
[image]
I will try some more extreme values on time constant and also look at increasing accelerometer noise and reducing range finder noise.

CarlOlsson, 2016-05-19:
RPL logs from wingwing with Pixracer. Altitude estimate is diverging in every flight

**master** 7f65e01d07ea0c1527ea1369750b8f4e53e86094
- [log1](http://logs.uaventure.com/view/EpfXekdBYBhAiuj6tnq3WQ) 

**master** but with sideslip fusion logging active. 
- [log2](http://logs.uaventure.com/view/Agu9hunwaRejuNkG2BX4S7) 
- [log3](http://logs.uaventure.com/view/Z5EeYCSvgasGo7atqMq6V)
- [log4](http://logs.uaventure.com/view/hTAeSR6DktCdZ4NCuNNa7L)

Here is a log from our system, pixhawk, hard mounted to the foam. Estimator did not diverge 
- [S100A_sess001](http://logs.uaventure.com/view/hRhAnJvNHjs5TWqo59LSUF)

Here is the same log but replayed if you have any problems working with the rpl log since it is generated by our code
- [S100A_sess001_replayed](http://logs.uaventure.com/view/PxeYhnjmXJFzXkAaE46efd)

priseborough, 2016-05-19:
The EKF and output observer vertical velocity in log http://logs.uaventure.com/view/ofn8G8zhNyyjJdaD3QvUN5
can be seen to have periodic 'kicks' of up to 0.2 m/s. Investigating this further showed it was caused by a problem with the delta velocity data (calculated from the IMU accelerations in the ekf2 module for this log). The following plot shows the z delta velocity which should be around -0.04 m/s, but instead has regular 'whiskers' where it drops to zero on one filter update and then goes to double the value the next frame. Some of the 'whiskers' are up to 0.2m/s in length which induces a 0.2m/s step in vertical velocity. This is causing problems with both EKF and output filter tracking.

[image]

What is puzzling is that the corresponding delta time value does not drop to zero:
[image]

This log is using the conversion from IMU gyro rates and delta angles performed in the ekf2 module, but the code is a simple trapezoidal integration performed here:

https://github.com/PX4/Firmware/blob/ekf2ImuSampleInvestigation/src/modules/ekf2/ekf2_main.cpp#L486

With the replay message written here:

https://github.com/PX4/Firmware/blob/ekf2ImuSampleInvestigation/src/modules/ekf2/ekf2_main.cpp#L874

I will investigate further tomorrow.

priseborough, 2016-05-19:
@CarlOlsson Your wing-wing has a bad case of vibration aliasing - see following plot of Z delta velocity when the filter has the first height divergence. You can see the beat frequency  reduce to DC as the vibration sweeps through a multiple of the sampling frequency. 

[image]

The following parameter changes work on replay allow it to survive the aliasing events without rejecting the baro:

EKF2_ACC_NOISE = 0.7
EKF2_BARO_GATE = 10

However you should re-work your isolation system as even with these revised parameters, the local and global position height is being affected by up to 3m and you have less margin to handle GPS errors. The objective of the isolation system is too ensure that the transmissibility transfer function has peaked and is rolling off below the Nyquist frequency and has plenty of attenuation by the time it gets to the sample frequency. A damping ratio of around 0.3 is ideal. If damping of the isolators is too high, then the attenuation with frequency is reduced, whereas for a material with lower damping it approaches 12dB/octave.

ndepal, 2016-05-20:
Through tuning, I am able to get good altitude tracking on the snapdragon.
I have significantly reduced the EKF time horizon by setting `IMU_BUFFER_LENGTH = 5` [here](https://github.com/PX4/ecl/blob/e3b9800cac459e7c279395c18191a872f465f848/EKF/estimator_interface.h#L237).
I have also drastically reduced `EKF2_TAU_POS` to 0.01, `EKF2_RNG_NOISE` to 0.01, and increased
`EKF2_RNG_GATE` to 20. These parameters are quite extreme and are not a long term solution to the problem.

[image]

Log here: http://logs.uaventure.com/view/bwv6GDn2JcrME4jWsWrd4c

Altitude control still leaves much to be desired, presumably due to the noisy z velocity estimate, on which the position controller is controlling.

Looking at the z velocity estimate, we see significant sharp jumps. These jumps occur in `LPOS.Z` before `EST0.s6`, which means that they come from the prediction with the IMU. These jumps cause audible spikes in the thrust when in alt ctl and prohibit tight tuning of the alt ctl gains.

[image]

Looking at `RPL1.aIz` and `IMU.AccZ`, we observe that `IMU.AccZ` almost exclusively has data points at opposing peaks. Spikes in `RPL1.aIz` occur whenever two data points in `IMU.AccZ` are 'on the same side' of the wave.

[image]

priseborough, 2016-05-20:
Can you give EKF2_IMU_MODE = 1 a try (it's the inbuilt integration from the IMU driver), whereas 0 is the alternative method I introduced that converts rates and accel data into delta angles and velocities in the ekf2 wrapper.

ndepal, 2016-05-20:
Here is a EKF2_IMU_MODE = 1 log: http://logs.uaventure.com/view/FtYH3i2MULLSwhptKfobvi

priseborough, 2016-05-20:
I have pushed some updates to https://github.com/PX4/Firmware/tree/ekf2ImuSampleInvestigation that provide an alternative method for the output prediction that gives a velocity and position that are kinematically more consistent with each other. The price is about 1.5% more CPU usage on Pixhawk and more noise on position as it currently unfiltered. To enable the new predictor, set EKF2_DERIV_FREQ = 3 as a starting value. Setting a negative number disables the new predictor.

The algorithm does a prediction forward from the fusion time horizon to the IMU time horizon by the length of the IMU buffer, every time the EKF does a prediction step. The velocity derivative and position are then processed using a second order complementary filter to calculate a position derivative, which is output as the local position velocity. The EKF2_DERIV_FREQ parameter sets the cross-over frequency. 

I have flown this branch with EKF2_DERIV_FREQ = 3 on the pixracrer and it was OK, but not as good as the default on my setup. Next week I will prototype and test some other methods to find the best balance between efficiency, prediction accuracy, kinematic consistency and fusion noise rejection.

ndepal, 2016-05-21:
We have placed a snapdragon on a QAV 250 frame (which was controlled with a PixRacer) to compare IMU noise levels. The Snapdragon was sitting on foam blocks, rather than hanging, and only had the power cable plugged in (i.e. without the TeraRanger, GPS/Mag, pass-through to PixFalcon that I usually have). In that setup, the IMU noise band is comparable to the PixRacer, +/- 1 m/s^2.

I also mounted the Snapdragon on top of the bebop frame, rather than hanging from below (with all 4 cables plugged in). In that setup, I have a noise band of about +/- 5 m/s^2.

Finally, I mounted the Snapdragon below the bebop frame, but rather than hanging it from foam blocks, I rest it on foam which in turn rests on L shaped brackets which attach to the bebop frame. This also results in a +/- 5 m/s^2 noise band. There might be slight room for improvement, as I quickly hacked this together with spare parts and tape. However, it seems to show that the damping properties of the foam we use are significantly worse in a hanging configuration compared to a more traditional setup. It also shows that the wind turbulence the Snapdragon is exposed to by being below the propellers is not a significant problem (at least compared to the foam block configuration).

With this new setup I tried out the new complementary filter mode in https://github.com/PX4/Firmware/tree/ekf2ImuSampleInvestigation

EKF2_DERIV_FREQ = -1: http://logs.uaventure.com/view/3CyvvE4rhoGnBaDGRJCti9
EKF2_DERIV_FREQ = 1: http://logs.uaventure.com/view/C9pvVzVRYb6B3PNPZiAxGD
EKF2_DERIV_FREQ = 3: http://logs.uaventure.com/view/zHWHqYwKR4V5E4wWx7UYyL
EKF2_DERIV_FREQ = 10: http://logs.uaventure.com/view/GiemygqfjdDxXdHdRiczia

Note that, for these logs, the tuning of the output complementary filter (`EKF2_TAU_POS`, `EKF2_TAU_POS`) and the IMU buffer length have been reset to the default values.

Qualitative observations:
With EKF2_DERIV_FREQ = -1 I still get occasional jumps in `LPOS.VZ` and `EST0.s6` (although less, due to the better IMU measurements), which cause spikes in thrust when in alt ctl. In general, altitude tracking is not too bad.

With  EKF2_DERIV_FREQ = 3 I do not get these spikes but altitude tracking seems worse than with EKF2_DERIV_FREQ = -1.

With EKF2_DERIV_FREQ = 1 altitude tracking is even worse and `EST0.26` and `LPOS.VZ` differ significanlty.

With EKF2_DERIV_FREQ = 10 when switching to position hold with flow, the quad starts to oscillate in the roll axis with increasing amplitude. After switching back to alt hold, this ceases immediately. In all other settings position hold with flow works without stability issues.

priseborough, 2016-05-22:
@ndepal The occasional jumps in LPOS.VZ and EST0.s6 are caused by what looks like processing or sampling errors in the delta velocity data . For that flight you had EKF2_IMU_MODE=1 so it was using the PX4 built in integrator. The following is a plot of the Z delta velocity data received by the ekf divided by the supplied time step. Note the 'spikes' on the data when the vehicle is at rest.

[image]

You could try EKF_IMU_MODE = 0 to use the ekf2 integrator method now I have changed the time-stamping method which will give us some more information on where to look for the problem. I will work through this with @julianoes.

TXS for the feedback on that output predictor prototype.

priseborough, 2016-05-22:
@julianoes @julianoes: Have you considered setting the MPU9250 sample rate divider in register 25 from 0 to 1 so that the gyro and IMU output are both at 4kHz? It will reduce the processing load in the IMU driver and  because the acceleration data is already sampled at 4kHz within the IMU wiht duplicate samples output, it will make no difference to aliasing resistance.

julianoes, 2016-05-22:
@priseborough: good idea, and it seems possible looking at the register map. I'll try this next week.

priseborough, 2016-05-23:
I have submitted a PR https://github.com/PX4/ecl/pull/148 that provides access to an optional method of correcting the output predictors velocity that uses position error. This ensures that the velocity is more kinematically consistent with the position which may provide better control loop performance under some conditions.

This method eliminates the noise sensitivity of the previous prototype in  https://github.com/PX4/Firmware/tree/ekf2ImuSampleInvestigation

julianoes, 2016-05-27:
@priseborough I tried to set the sample rate divider to 1 but I didn't see any effect on the number of samples waiting in the buffer to be emptied.

Looking at the register map I realized that the option is not available for 8 kHz sampling.

> Divides the internal sample rate (see register CONFIG) to generate the
> sample rate that controls sensor data output rate, FIFO sample rate.
> NOTE: This register is only effective when Fchoice = 2’b11 (fchoice_b
> register bits are 2’b00), and (0 < dlpf_cfg < 7), such that the average filter’s
> output is selected (see chart below).
> This is the update rate of sensor register.
> SAMPLE_RATE= Internal_Sample_Rate / (1 + SMPLRT_DIV)

FCHOICE_B is 0b00 so FCHOICE is 0b11, that's ok.

However, DLPF_CFG is 0 because we want 8kHz sample frequency (BW 250Hz). If we set it to 1, we only get 1 kHz sample frequency (184 Hz BW).

priseborough, 2016-06-16:
@ndepal, @CarlOlsson , @julianoes my understanding is that this issue has been addressed through a combination of tuning changes, enhancements to the output observer and fixes to hardware in some instances. If not can you please provide a replay log. TXS.

CarlOlsson, 2016-06-16:
I agree, for me changing the mounting and the drivetrain eliminated the problem

ndepal, 2016-06-17:
Yes with the new damping method the performance is quite good. I still observe some small jumps which appear to be caused by IMU sampling issues discussed above. But as this is not an ecl problem, I would consider this issue resolved

julianoes, 2016-06-17:
@ndepal: I haven't seen this spikes myself recently. Is this with latest master? Can you send me a logfile?

ndepal, 2016-06-20:
No, that was actually a few weeks ago, you're right. I can't test latest master on snapdragon right now, but I'll let you know if I encounter it again.

tumbili, 2016-10-06:
@priseborough Closing this as well, probably outdated.

closed 2016-10-06 by tumbili