PX4/PX4-ECL #141: Oscillations and settling time of EKF using EV

pickledgator, 2016-05-19:
Ran some more motion tests today using the EV fusion updates that were pulled into master a few days ago (and the new alignment PR from yesterday -- alignment seems much faster). I'm passing in vision updates at 100Hz and they are pretty clean. There's quite of a bit of overshoot and settling time in the ekf (as compared to the vision measurements) that I'm struggling to tune out. Eg., as I'm walking with the drone, when I physically stop motion, the vision measurements flatline, but the ekf oscilates twice or so and eventually settles. I've played around with the noise parameters for both EV and accel/accel bias, with no clear trends to reduce the overshoot. Not sure what other nobs I can turn. Could use some advice. Log file with replay enabled here (not sure why the LPOS data isn't show up on logmuncher?): http://logs.uaventure.com/view/SWMESe69SkMENaTSPwHwcZ

Thanks!

priseborough, 2016-05-23:
I can't get you log to work on the replay. Will investigate further.

priseborough, 2016-05-23:
It's because https://github.com/PX4/Firmware/pull/4382 was not finished. If you want help with the EV feature, then you must get that PR cleaned up, tested and pulled into master. @LorenzMeier for info.

pickledgator, 2016-05-24:
Thanks, @priseborough. Will get an updated log out when #4382 is resolved.

priseborough, 2016-05-25:
@pickledgator Can you please get a replay log using the following branch and send it too me as @devbharat is currently unable to generate logs. This will help me get #4382 resolved.

https://github.com/priseborough/Firmware/tree/px4_push_ev_Ekf2_replay

TXS

pickledgator, 2016-05-25:
@priseborough I pulled in your branch compiled and flash, but apparently my logs are still corrupt. They open in FlightPlot, but are missing tons of fields like, LPOS but have fields like attitude. When the px4log is loaded, it says there are errors but I have no idea how to troubleshoot this. My process was to pull down PX4/master, then merge in your branch and add the few lines of modifications that I need to get my vision data piped in. All data shows up fine live in QGC, but somehow isn't being logged correctly. Any help is appreciated so that I can get you some data.

devbharat, 2016-05-25:
Are you talking about the replay log ? I does not log LPOS etc, only replay stuff (test RPL1.t) and attitude. So that is OK.

priseborough, 2016-05-25:
I think I know what the problem could be - the changes to the field descriptor for the RPL5 message did not make it into the PR. I will fix that now.

pickledgator, 2016-05-25:
Pulled in the change and collected a log. No errors when running the posix_sitl_replay. Changed PRIO_BOOST to 3 for this dataset. AID_MASK set to 24 -- QGC shows lpos converging to vision, still feels like the logs are missing data. Using SD card that came with pixhawk, so assuming that is sufficient.

https://www.dropbox.com/s/bqdkoi62x3j4ddi/log001.px4log?dl=1
https://www.dropbox.com/s/cymfzflsjcivofc/replay.px4log?dl=1
https://www.dropbox.com/s/88lrtko8eqfwi72/replay_replayed.px4log?dl=1

priseborough, 2016-05-25:
OK, so no RPL5 messages were extracted and there are lots of warnings about bad bytes in the log . I found a bug in the replay program which has been fixed so have pushed this and another minor correction, but neither explain the bug. @tumbili are you able to do a review of the RPL5 message logging to see if you can find anything?

pickledgator, 2016-05-25:
I'll dig in a bit more in the morning with some console prints to see if I can track down the lack of RPL5 data.

pickledgator, 2016-05-25:
Ah! Found it. Check the enum numbering here: [sdlog2_messages.h#L561-L623](https://github.com/priseborough/Firmware/blob/27ad3cd3d1d67991d942edf265afa10440c746e1/src/modules/sdlog2/sdlog2_messages.h#L561-L623). There are multiple of the same number. I'll let you fix it in your PR so we don't have to do two merges. Fixing the enums results in px4logs with no errors, and the RPL data is there.

priseborough, 2016-05-26:
pushed the fixes

priseborough, 2016-05-26:
Can you please record some more data using the updated branch. The enum numbering used for the logging and replay have to be consistent.

pickledgator, 2016-05-26:
This should have the correct enums from your PR. 
http://logs.uaventure.com/view/w9UrF6JWWbQageMVdp4mZ3

priseborough, 2016-05-27:
The replay results using your data with https://github.com/PX4/Firmware/pull/4647 were excellent. I will enable adjustment of the position noise via parameter given it is not yet supported over mavlink and then I think it should be good to pull in.

[image]
[image]
[image]
[image]

pickledgator, 2016-05-27:
Awesome, looking forward to pulling in the updates and running a test flight.

pickledgator, 2016-06-01:
Hoping to run a full flight test of the new EV changes sometime this afternoon. Will provide log after collected.

priseborough, 2016-06-16:
@pickledgator what is the status on your testing? If it was successful, then we can close this - if not can you please provide replay logs. TXS.

pickledgator, 2016-06-16:
Had a crash on inav (with corrupted log unfortunately since we had to pull power), so in process of hardware rebuild. Hope to be able to run another ECL flight log by the end of next week.

priseborough, 2016-10-26:
Unable to reproduce. Closed due to lack of follow-up information from the OP.

closed 2016-10-26 by priseborough