PX4/PX4-ECL #172: EKF: come up with recommended default tuning parameters for lower quality boards

priseborough, 2016-07-07:
This user log with a pixhack board had  large vibration errors on the accelerometer and very noisy baro data that caused a temporary loss of height reference. A different set of tuning defaults would have prevented this.

Note: The internal isolation design of the pixhack is poor and can cause clipping if resonance of the mount occurs. Unfortunately there are lots of those boards out there and users do not think they need to be isolated.

http://logs.uaventure.com/view/DJwKnqsFtHVJQuaKoEK8UY

This particular log would likely require a  increase in baro observation noise to 3.0m  and accel process noise to 0.5 m/s/s which would come at a cost of reduced angular accuracy and increased susceptibility to GPS glitches.

LorenzMeier, 2016-07-07:
@priseborough Can share the exact param names for these?

priseborough, 2016-07-07:
EKF2_ACC_NOISE = 0.5
EKF2_BARO_NOISE = 3.0

priseborough, 2016-10-26:
If we can collect logs using the new ulog format for representative boards, then I now have a python script tool that will generate population statistics to assist in determining appropriate values. 

https://github.com/priseborough/pyulog/tree/ekfAnalysis-wip