PX4/PX4-ECL #173: Handle dual GPS in ekf2

CarlOlsson, 2016-07-07:
@LorenzMeier @priseborough 
Dual GPS is now supported in PX4/Firmware but not in EKF2. Here is a plot of a switch of module.
[image]
We should specify how to handle it and we would be happy to work on it.

priseborough, 2016-07-07:
It is the jump in position that is the problem. One option is to apply an offset to the GPS when you switch so that it matches the previous active unit (if available) or the EKF position at the time of the measurement (need to allow for GPS delay). That position offset could then be decay radially to zero at a slow rate rate, eg 0.1 m/s.

CarlOlsson, 2016-08-03:
@priseborough this should be easiest to handle outside of the ecl library, do you agree?

priseborough, 2016-08-03:
Agree. My preference is to perform any sensor switchover outside the library. This would include other examples of dual sensors (Baro, airspeed, etc).

CarlOlsson, 2016-08-27:
@LorenzMeier I implemented this for us and it works well, plots from switching GPS module twice while airplane is stationary on the ground. Test ratio > 1 corresponds to GPS being rejected by the EKF.
[image]
[image]
I can push it upstream if it is interesting for you but I am not sure how to decide about when to switch module in a generic way, maybe we could add a field to vehicle_gps_position of which module is in use?

priseborough, 2016-08-27:
UBlox receivers provide a number of quality metrics that can be used for selection logic (horizontal accuracy, speed accuracy, etc).  Speed accuracy is usually an early indicator of data quality problems, whereas the horizontal and vertical position accuracy figure takes longer to degrade. 

If your solution involves decaying a switchover position offset correction offset to zero, it may help to offset correct the reported velocity for rate of position offset decay.

CarlOlsson, 2016-08-29:
Thanks Paul. This is indeed very interesting and we should definitely look into it at some point, right now we base the switching logic only on regime (hover/cruise). However, what I meant was more focused on in which module to switch GPS and how to publish it. Maybe that is indeed a bigger question on how to handle redundant sensors in general such that every module doesn't have to decide on its own. If the best place to have the logic is not in the EKF then I can push the code right away and base the switching on a boolean coming from somewhere, if not then we could start looking into the best switching logic right away.

priseborough, 2016-10-26:
I am closing this and have re-opening the issue in PX4/Firmware https://github.com/PX4/Firmware/issues/5759

closed 2016-10-26 by priseborough