PX4/PX4-ECL #179: EKF: airspeed fusion and wind estimate

dagar, 2016-07-28:
I'm interested in using EKF2 for fixed wing flights, but I see airspeed fusion is currently disabled. What needs to be done to enable airspeed fusion? I can help with testing and debugging.

https://github.com/PX4/ecl/blob/eaae95fdc4e48e3946a463941dfb9d5e495e975f/EKF/control.cpp#L744

CarlOlsson, 2016-07-29:
Hi @dagar great that you want to start using airspeed fusion and can help with debugging!
We have been flying with airspeed activated for many weeks and it was all fine until we hit a corner case a couple of days ago where the wind states diverged. @priseborough helped track down the problem to how the wind state variances are currently initialized. This have to be changed. Regarding [synthetic sideslip fusion](https://github.com/PX4/ecl/pull/178) I am currently trying to get [replay working again](https://github.com/PX4/Firmware/issues/5143) in order to further test it. If you have access to a stable test airplane it would be really helpful if you could make a replay testflight with some code I can prepare

tumbili, 2016-08-12:
@dagar I think #183 would be ready for a test flight. If it's possible then it would be nice if you could enable replay EKF2_RPL_EN=1 so that we can track down any remaining issues with replay.

dagar, 2016-08-12:
I've never tried ekf2 replay logging, can px4fmu-v2 handle it?

tumbili, 2016-08-12:
@dagar yes it can! Make sure not to use the shittiest SD card you find.

priseborough, 2016-10-26:
Airspeed fusion is working: https://github.com/PX4/ecl/pull/183

closed 2016-10-26 by priseborough