PX4/PX4-ECL #181: ECL_PitchController body_fixed_turn_offset issue

dagar, 2016-08-11:
The ecl pitch controller body fixed turn offset is an additional term added to the rate setpoint. 

https://github.com/PX4/ecl/blob/master/attitude_fw/ecl_pitch_controller.cpp#L153

    float body_fixed_turn_offset = (fabsf((CONSTANTS_ONE_G / airspeed) *
				  		tanf(constrained_roll) * sinf(constrained_roll)));

With large roll angles this term becomes enormous and causes problems like in this recent flight. Note the small pitch setpoint, but full elevator deflection.

[image]

LorenzMeier, 2016-08-11:
I guess that sounds a reasonable start. Or the value it would assume at 30 or 40 degrees roll and lock it there.

tumbili, 2016-08-11:
@dagar Can you please post the entire log?

dagar, 2016-08-11:
What about constraining the roll it uses for the offset calculation to +- roll setpoint? Then the offset is based on the planes actual roll, but will never go beyond the roll limits, and will only be active during a commanded turn. I need to think through the inverted logic.

mujahid-abdulrahim-p, 2016-08-11:
The issue here is that the body_fixed_turn_offset term attempts to compensate for the tilted lift vector and prevent altitude loss at large roll angles.  The term is useful for relatively small angles, but beyond some roll angle, the appropriate response might be to stop trying to maintain altitude and allow the aircraft to recover to a normal operating condition.  Even for aircraft that can fly knife edge, it's unlikely that an aircraft should fly a coordinated turn at bank angles beyond 60 degrees.

dagar's comment about the roll setpoint is appropriate, since this provides some notion of autopilot intention and does not apply a pitch rate feedforward in response to a roll disturbance.

PR #191 (2016-10-06): ecl attitude_fw fix excessive body fixed turn offset
-limit body fixed turn offset contrained roll to possible
roll setpoint range
-fixes #181
commit 0d02bb26 (2016-10-06): ecl attitude_fw fix excessive body fixed turn offset

-limit body fixed turn offset contrained roll to possible
roll setpoint range
-fixes #181
closed 2016-10-06 by LorenzMeier