PX4/PX4-ECL #185: ECL_L1_Pos_Controller::navigate_loiter fly to center

dagar, 2016-08-12:
Is there a reason l1 navigate_loiter always navigates to the center first before joining the loiter? It seems like it would be much cleaner to directly enter the loiter and avoid the huge awkward turn.
Just checking before I start changing something I don't yet fully understand.

LorenzMeier, 2016-08-12:
I think we should indeed target the loiter radius instead right away (setting the acceptance radius as loiter + acceptance radius should do that already, no?)

LorenzMeier, 2016-08-12:
@dagar Should IMHO be a simple fix based on acceptance radius

dagar, 2016-08-22:
I gave that a quick try adding distance to the waypoint (center) as another condition for the crossover.
https://github.com/PX4/ecl/blob/master/l1/ecl_l1_pos_controller.cpp#L292
Most of the time it works as intended (an improvement), but occasionally I saw some weird behavior turning the wrong way I'm not comfortable with.

LorenzMeier, 2016-11-05:
I'm closing this since the behaviour you saw seems fine. We might want to tune some of the tresholds by e.g. entering the loiter earlier than you currently have.

closed 2016-11-05 by LorenzMeier