PX4/PX4-ECL #194: Altitude estimate is drifting

CarlOlsson, 2016-09-23:
Here is another <4 min flight where the baro didn't drift at all but the altitude estimate drifted 7 meters.
[Logfile](http://logs.uaventure.com/view/DmABoaBZ8g39aqSvBQpBc7)

CarlOlsson, 2016-09-23:
This was caused by hardware failure and altitude "drifted" after impact.

closed 2016-09-23 by CarlOlsson

tumbili, 2016-09-23:
@CarlOlsson I think the effect you are seeing is due to the effect that the vehicle hit the ground at the end.
If you look at your baro innovations then they don't showy any long term bias, so I don't think this is a problem.
[image]

tumbili, 2016-09-23:
@CarlOlsson What do you mean by hardware failure?

CarlOlsson, 2016-09-23:
Yes in this case it was a bit difficult to say how much the altitude drifted but what do you think of the log file linked here:
https://github.com/PX4/ecl/issues/44
Why does LPOS_Z drift more than the baro?