PX4/PX4-ECL #205: EKF position resets

CarlOlsson, 2016-10-24:
Did three flights today with master ecl 65de4962b5fd45bbdce9d7ef84ef486c406d8e2d
[flight1](http://logs.uaventure.com/view/Nsdwvw6YVTa2bFPYnWb3AD)
[flight2](http://logs.uaventure.com/view/9oDo8aE4KuJcZebrZqhcRH)
[flight3](http://logs.uaventure.com/view/6r4fkqDtUeAB9i2Lq7322g)

In the first flight the estimator diverges completely and several position resets occur.
In the second flight the airspeed fusion was turned off for half the flight then activated, but airspeed was rejected for a large portion of the flight.
Third flight looks better but still very bad airspeed innovations.

I suppose the problems come from a bad airspeed calibration or sensor, but I am surprised how much it couples into the rest of the estimator if that is the case

priseborough, 2016-10-24:
The airspeed accuracy was very poor and airspeed reads zero for significant parts of the flight:

[image]

Bad airspeed measurements can affect the states in the same way that bad GPS velocity measurements can. In this instance you have a sensor that is regularly telling the vehicle it is stationary relative to the wind and a wind process model that only allows for speed variation at a specified rate. 

Adjustments to tuning would yield reduced sensitivity of the velocity and position estimates to a similar error, but that would come at the cost of a degraded wind estimate with good airspeed sensing. Another approach is to set the Kalman gains for the non windspeed states to zero if some other form of aiding (eg GPS) is still available. I will work on a patch to do this.

priseborough, 2016-10-24:
See https://github.com/PX4/ecl/pull/206

closed 2016-10-26 by priseborough