PX4/PX4-ECL #218: GPS velocity glitch feeding into estimation

LorenzMeier, 2016-12-13:
@priseborough This GPS velocity glitch is feeding into the estimation pipeline (it has no position glitch associated it seems). Is this something you feel we can guard against?

http://review.px4.io/plot_app?log=4624b22c-2e0d-42f3-ae82-7f56645cbbf0

priseborough, 2016-12-13:
We already have innovation consistency tests for GPS which were operating for this flight. Here are the normalised test ratios and Innovation plots for the various observations. Inspection of data shows that the significant flight event was  not due to a GPS velocity glitch.

1) An inertial sensing error at 260 seconds drags the height estimate away from the Baro and GPS an requires a reset. This is usually distinguishable from a  GPS or Baro error because the sign of the vertical velocity and position innovations is the same. This vehicle has a vibration problem that needs to be fixed.

 2) Other than one short duration event at around 190 seconds, GPS innovations levels in the horizontal are not statistically large enough to indicate a glitch. However the default innovation test level for velocity could be reduced to 3.0 for earlier rejection of GPs velocity errors which has been found to work well on ArduCopters EKF. Note that this would not have changed the flight experience for this user.

3) This vehicle has high levels of GPS velocity noise throughout flight and requires improvements to the GPS installation if it is to fly well.

My recommendations are that the default value of EKF2_GPS_V_GATE be reduced from 5.0 to 3.0 and that the user reduce their board vibration and GPS noise.

[image]

[image]

[image]

[image]

LorenzMeier, 2016-12-13:
Thanks! The GPS signal is multi path below trees. Nasty, but not totally uncommon. Can you send a PR for the default param change?

priseborough, 2016-12-13:
Will do

priseborough, 2016-12-13:
Yes the surrounding trees would have reduced the SNR and increased velocity noise, however a good GPS installation would normally not be bothered by that level of obscuration.

priseborough, 2017-07-18:
Loss of height estimation due to bad vibration mitigated by https://github.com/PX4/ecl/pull/267

closed 2017-07-18 by priseborough