PX4/PX4-ECL #222: Addition of FLT_EPSILON reference has broken the Snapdragon build on master

jywilson, 2016-12-20:
@LorenzMeier 
After the following commit:
https://github.com/PX4/ecl/commit/6f78ef408425d5b9bc7e0eaceb1af9e5fd73b424

The build for the Snapdragon Flight board produces the following errors:

/local/mnt/workspace/jenkins-slave/workspace/PX4_External_CI/src/lib/ecl/validation/data_validator_group.cpp:154:23: fatal error: use of undeclared identifier 'FLT_EPSILON'
) && (confidence > FLT_EPSILON)) {
^

1 error generated.

LorenzMeier, 2016-12-21:
Thanks. Fixed on 1e625d024cb25842e77c5c88959bea6bdda28350. We need to get Snapdragon back into CI. @dagar I believe there was a specific blocker to that?

closed 2016-12-21 by LorenzMeier

jywilson, 2016-12-21:
There is still one issue.  Nothing major.  I filed it here:
https://github.com/PX4/Firmware/issues/6125