PX4/PX4-ECL #225: ECL verbosity

LorenzMeier, 2016-12-26:
@priseborough It would be nice if ECL could be less verbose on boot. Changing from:

```
INFO  [lib__ecl] EKF IMU buffer length = 21
INFO  [lib__ecl] EKF observation buffer length = 16
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
```
to:
```
INFO  [lib__ecl] EKF aligned (pressure height, IMU buf: 21, OBS buf: 16)
```

You can still keep any error messages. We're following the POSIX design pattern: No news is good news and components should offer as little positive notifications as possible.

On the flip side *every* error must be handled and reported. Taking this principle to the extreme does show errors much more easily as they stand out and people avoid reporting errors by not reporting success, which is a terrible pattern.

nicolaerosia, 2017-07-10:
@LorenzMeier I believe this can be closed

LorenzMeier, 2017-07-10:
Thanks! And I will work hard to get your PR's in this week after we've released 1.6.4

closed 2017-07-10 by LorenzMeier