PX4/PX4-ECL #240: Cmake lists error

FarmerLjx, 2017-02-17:
Open as a CLion project, and get this output by Cmake:

CMake Error at CMakeLists.txt:37 (px4_add_module):
  Unknown CMake command "px4_add_module".

psiphi75, 2017-02-21:
I get the same issue.  I just want to compile the files in `elc/EKF/tests/base/`.  I run 'cmake ./` from that folder and get:

```sh
CMake Error at CMakeLists.txt:33 (px4_add_module):
  Unknown CMake command "px4_add_module".
```

dagar, 2017-02-21:
The current top level CMakeLists.txt is used for including ecl in PX4 Firmware. If you want to build EKF2 standalone use EKF/CMakeLists.txt or run build.sh.

FarmerLjx, 2017-02-23:
build.sh worked for me, but made some changes to eliminate compilation errors.

diff --git a/EKF/CMakeLists.txt b/EKF/CMakeLists.txt
index 0d7d6a9..cb3b107 100644
--- a/EKF/CMakeLists.txt
+++ b/EKF/CMakeLists.txt
@@ -66,6 +66,8 @@ set(SRCS
        mag_fusion.cpp
        gps_checks.cpp
        airspeed_fusion.cpp
+       optflow_fusion.cpp
+       terrain_estimator.cpp

dagar, 2017-02-23:
@FarmerLjx can you open a pull request?

FarmerLjx, 2017-02-27:
It is just add two source files to the *SRCS*, so really need a pull request? I'm still learning how to use my data to test the effect of EKF.

dagar, 2017-02-27:
That's how anyone, including developers with write access to the repository
would make this small change.

closed 2017-02-27 by dagar

FarmerLjx, 2017-02-27:
Ok, i see, thanks!

PR #243 (2017-02-27): cmake add all EKF sources
 - update matrix to latest (matches PX4)
 - trivial cmake cleanup
 - travis-ci build with both gcc and clang
 - closes #240
commit eb4f5ecb (2017-02-27): cmake add all EKF sources (#243)

 - trivial cmake cleanup
 - travis-ci build with both gcc and clang
 - update matrix to latest (matches PX4)
 - closes #240