PX4/PX4-ECL #249: How attitude is corrected by accelerometer in EKF2

Gangti, 2017-03-10:
Hi, all

I'm currently trying to understand EKF2 implementation. I found yaw angle could be corrected by magnetometer, but cannot understand how pitch/roll angle is corrected by accelerometer, for accelerometer isn't taken as an observervation.
However, I've noticed that following code seems related to pitch/roll angle correction, for when I commented it out, the pitch/roll angle drifting slowly.
ecl/EKF/vel_pos_fusion.cpp
```
83:	if (_fuse_pos) {
...
89:		if (_control_status.flags.gps) {
...
101:		} else if (_control_status.flags.ev_pos) {
...
110:		} else {
...
117:			_vel_pos_innov[3] = _state.pos(0) - _last_known_posNE(0);
118:			_vel_pos_innov[4] = _state.pos(1) - _last_known_posNE(1);

```
Anyone can explain the logic briefly? Appreciate it.
@priseborough 

----

priseborough, 2017-03-10:
It's hard to explain briefly. The EKF compares the velocity velocity and position predicted by the inertial navigation calculations with the GPS measurements and is able to use the inconsistency to estimate the error in the attitude. I recommend you study the matlab derivation and take a course in Kalman filtering theory and inertial navigation if you wish to understand it. 

https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m

closed 2017-03-10 by priseborough

Gangti, 2017-03-12:
@priseborough
Thanks for your explaination. I'll study related materials.
BTW, it seems that GPS is a necessity for correction of attitude, velocity and position.
Only with MAG, no GPS, for EKF2, I think yaw will be corrected, and velocity/position will drift. but will attitude(roll/pitch) drift of not?
Maybe I should disconnect GPS in Pixhawk, and take a look of the EKF2 output.

priseborough, 2017-03-12:
Without GPS, the filter operates in a mode where it uses position observations at the last known position. This constrains roll pitch drift errors provided the vehicle is not doing sustained horizontal manoeuvres.

Gangti, 2017-03-13:
@priseborough 
Thanks for your reply. It does helps to understand EKF2.
Without GPS, as the comment of vel_pos_fusion.cpp says "No observations - use a static position to constrain drift", if the vehicle does standstill, the attitude estimation should be correct. Just as the condition you mentioned "provided the vehicle is not doing sustained horizontal manoeuvres".
In other words, if the vehicle is moving constantly, the attitude estimation will slightly be wrong for a moment, and after it stops moving, the attitude estimation will be right again.
Is the understanding right?

Just now, I unplug the GPS and blade, and take the vehicle horizontally in hand far away, meanwhile monitor the attitude output, the pitch/roll angles does go over 1 degrees. Finally, I lay the vehicle horizontally on ground, the pitch/roll angles become 0 degree in a moment.
Can this experiment show the EKF2 does slight wrong attitude estimation with GPS, and in a sustained horizontal moment?

Thanks again for you patience.