PX4/PX4-ECL #271: EV/Mocap Local Position Incorrect

dennisss, 2017-05-10:
I am running running with the latest PX4 master modified very slightly to use the att_pos_mocap mavlink message for external vision input to EKF2 (https://github.com/dennisss/Firmware/blob/tansa-next/src/modules/ekf2/ekf2_main.cpp#L758)

I've configured the parameters as:
> param set EKF2_AID_MASK 24
> param set EKF2_HGT_MODE 3
> param set EKF2_EVP_NOISE 0.001
> param set EKF2_EVA_NOISE 0.001

When running in hover using an external controller, all the local position axes diverge randomly (all these lines should be flat)

[image]

[image]

Here is the flight log: http://logs.uaventure.com/view/rL77t6PCvJDVgkKYAemFGP

Hardware is a PixRacer. In real life, the drone stays in one spot the entire time. This works just fine when switching to LPE/ATTQ

Any ideas on why the position is the wrong values?
Thanks!

dennisss, 2017-05-10:
Nevermind, hardware issue

closed 2017-05-10 by dennisss

broody, 2017-10-06:
Hi @dennisss, just to confirm, were you able to fly successfully with this patch? MOCAP with EKF2.

dennisss, 2017-10-06:
@broody yes. Mocap works very well with the patch.

mzahana, 2017-12-14:
@dennisss Is this merged into PX4 master?

mandelyoni, 2018-08-10:
Hi Dennisss,

I cloned your PX4 version and tried to build it but received errors. 
Any suggestion would be appreciated.

Thanks,  Yoni

mzahana, 2018-08-10:
@mandelyoni In case this helps, PX4 can fuse MOCAP in EKF2. Check my comment in the following issue,
https://github.com/PX4/Firmware/issues/6364#issuecomment-410481134

mandelyoni, 2018-08-10:
Great!
Thank you very much  :)

mandelyoni, 2018-08-10:
@mzahana , Do you know if anyone has tried it with the Intel Aero drone?

mzahana, 2018-08-10:
@mandelyoni No, I don't. But I don't see a reason why it shouldn't.

mandelyoni, 2018-08-10:
Ok, we'll try it and update. 
Thanks

mandelyoni, 2018-09-08:
Hi,

We managed to operate the Intel Aero with the Optitrack Mocap, according to
https://github.com/PX4/Firmware/issues/6364#issuecomment-410481134

Thank you

revantad, 2018-10-17:
@mandelyoni Hey, I am working on the same platform and would definitely appreciate some pointers.

The mocap data is being published to the '/mavros/vision_pose/pose' and the data is currently being displayed in QGC. Also, the 'EKF_AID_MASK', 'EKF2_HGT_MODE' and 'EKF2_EV_DELAY' parameters are changed to appropriate values but the local_pose topic echos only Z and Orientation data but doesn't echo X and Y data.

Any help is appreciated.

potato77, 2018-10-21:
@revantad Do you fix this problem? I also get the same problem.

mandelyoni, 2018-10-21:
Hi,

Did you download the latest PX4?

potato77, 2018-10-22:
I am also using LPE. I fix the problem of xy estimation using the lastest firmware. But the yaw angle does not work.

| |
fatmoonqyp
邮箱：fatmoonqyp@126.com
|

Signature is customized by Netease Mail Master

On 10/22/2018 12:47, Revant Adlakha wrote:

@potato77 Hey, I actually am using the older version of Pixhawk that still supports LPE and I am using /mavros/mocap/pose to publish the pose. Do let me know if you get a fix for the EKF2 situation. I am actually facing a local take off problem, let me know if you could help me with that.

revantad, 2018-10-22:
@potato77 Maybe you want to change the `ATT_EXT_HDG_M` to Motion Capture