PX4/PX4-ECL #284: Height reset redundant statement

ChristophTobler, 2017-06-20:
In https://github.com/PX4/ecl/blob/7f1cca5b28087055f8343ec069852bedd3b86cd7/EKF/control.cpp#L483, the first statement is always true since `_params.hgt_reset_lim = 0.0f` (https://github.com/PX4/ecl/blob/7f1cca5b28087055f8343ec069852bedd3b86cd7/EKF/common.h#L231 and there is no parameter that the user could set)

Wouldn't it make sense to compare `P[9][9]` with the IMU variance?
And if the first statement works, shouldn't it be
```
(P[9][9] > sq(_params.hgt_reset_lim)) || hgt_fusion_timeout || continuous_bad_accel_hgt
```
as we also want a reset if `P[9][9]` grows to big independent of the timeout?

priseborough, 2017-11-15:
This is leftover from a previous implementation for the reset logic that attempted to use growth in the variance to trigger a reset. This was abandoned because the main cause of resets has been clipping of accelerometers due to vibration which causes extreme error growth that is not predicted correctly by P[9][9].  I will remove this test and the associated parameter.

priseborough, 2017-11-15:
If P[9][9] becomes very large before the timeout period, then the innovation consistency check will pass and the data will be fused anyway.

closed 2017-11-15 by priseborough