PX4/PX4-ECL #287: Sideslip based Wind Estimation

JohannesFriis, 2017-06-23:
It seams there is a problem with wind estimation without airspeed sensor. The estimated wind speed is very close to zero and in general 1/10 - 1/20 of what I would have expected. The problem is observed when flying PX4 tag v1.6.3 on a AERT FW (twinstar) using the setup described in https://github.com/PX4/ecl/pull/207 The controller is a pixhawk.
Last known working setup for wind estimation is PX4 tag v1.5.5

[08_16_30.zip](https://github.com/PX4/ecl/files/1097375/08_16_30.zip)

Best regards
Johannes Friis
jf@sky-watch.com

dagar, 2017-06-23:
Log on flight review - http://logs.px4.io/plot_app?log=44333367-d383-4491-a7c8-2fae5be8f67d

EKF log process output (./Tools/ecl_ekf/process_logdata_ekf.py)
[08_16_30.ulg.pdf](https://github.com/PX4/ecl/files/1098011/08_16_30.ulg.pdf)

JohannesFriis, 2017-06-26:
Thanks for running my log.

I am still confused why the wind_estimate topic logs so little wind. I believe there was a great deal more wind that day than what was logged

[image]

JohannesFriis, 2017-06-29:
Hi @dagar and @priseborough  

I Have been doing some code comparison with older ECL versions to try to locate the change that causes loss of a valid wind estimation. as described above.

I Have established 2 critical changes so far

- 10 Apr commit 9f48c0505bb093cabbb1b8818131422eb4dc0ed0 
EKF/covariance.cpp l.207 changes a threshold for wind velocity state variances. from 1000 to 2.

Have this been tested without use of a Airspeed sensor?

- on 10 Apr commit 465b14592961bc174d4d534063223fc66f41532b
introduction of drag fusion
EKF/control.cpp l. 876: _control_status.flags.wind = false; if _params.fusion_mode & MASK_USE_DRAG is false. 
I am under the impression that drag fusion does not apply to FW? but drag fusion forces _control_status.flags.wind false when it is not used.

If I reset the wind velocity state variances to1000, and outcoment drag fusion I again get a nice smooth wind estimate without use of an airspeed sensor.

[image]

Best Regards 
Johannes Friis
jf@sky-watch.com

priseborough, 2017-06-29:
I will submit a fix for this today. I will to look at the auto tester to see how these defects introduced  passed fixed wing SITL testing.

priseborough, 2017-06-29:
@JohannesFriis  Would you be able to provide a  log gathered using SDLOG_MODE=1 and EK2_REC_RPL=1 ? If you use a high quality U3 SD card (eg Sandisk extreme) then logs can be gathered without any loss of data. I can use it to test the changes on replay.

 I do not have a fixed wing platform available for PX4 testing and SITL has a time-slip problem running on my computer.

dagar, 2017-06-29:
EKF2_ARSP_THR defaults to 0 and isn't being changed for FW (including SITL tests).
EKF2_FUSE_BETA is also always off.

Should we do a pass and make sure the testing is running with reasonable EKF settings?

Have you tried running gazebo headless? 

    HEADLESS=1 make posix_sitl_default gazebo_standard_vtol

priseborough, 2017-06-29:
No I haven't tried the headless mode.  How do I test a fixed wing (non VTOL) flight that is representative of a hand or catapult launch?  It is important that we get the same immediate transition from stationary into forward flight.

I also think there is more going on with the OP's log. The GPS velocity is rejected immediately on launch before any sideslip observations are fused which combined with the 1 radian offset between vehicle yaw and GPS ground course suggest the failure was primarily the result of a bad compass.

priseborough, 2017-06-29:
[image]

priseborough, 2017-06-29:
I think the reduction in initial wind uncertainty combined with the sideslip fusion has made the filter more vulnerable to initial yaw errors. I will make the initial uncertainty a parameter so it can be tuned for different applications, but the long term fix is to do what ArduPilots EKF's do for fixed wing which is if large GPS velocity innovations are detected on launch, the vehicle yaw is realigned to reduce the innovations to zero and the magnetic field estimates are reset.

priseborough, 2017-06-29:
I found the logic bug introduced by the drag fusion that is preventing proper wind estimation for fixed wing in Ekf::controlDragFusion().

dagar, 2017-06-29:
I'm not sure about simulating hand or catapult launch in gazebo, but the current gazebo plane model is probably unrealistically powerful.

    HEADLESS=1 make posix_sitl_default gazebo_plane

Beyond that please let me know if you think it's valuable to simulate something faster and I'll figure out how to give it an initial velocity.

priseborough, 2017-06-30:
I found another instance of this bug here: http://review.px4.io/plot_app?log=a385efa2-060b-43be-aa76-9b2244aff892 affecting a fixed wing vehicle with good compass.

JohannesFriis, 2017-06-30:
Hi @priseborough 

i Have tried to generate the log file that you requested. I have no prior experience with EKF replay so I hope the log contains the info you need. The firmware is still tag v1.6.3 and SDLOG_MODE=1 and EK2_REC_RPL=1 as you requested. The best SD card I had available was a 32GB samsung UHS1 pro plus card. I Hope it is fast enough. 

Is there a more extensive guide to EKF replay than https://dev.px4.io/en/log/ekf2_log_replay.html I Would like to read/learn/try it over the weekend? 

I will not be able to do test flights over the weekend but if needed I will gladly retest again next week.

Best Regards
Johannes Friis
jf@sky-watch.com
[08_27_18.zip](https://github.com/PX4/ecl/files/1114734/08_27_18.zip)

priseborough, 2017-06-30:
That EKF replay method has been deprecated and I have requested that the corresponding documentation be removed. Please follow the EKF2 replay section here: https://dev.px4.io/en/debug/system_wide_replay.html#ekf2-replay

bkueng, 2017-07-03:
> That EKF replay method has been deprecated and I have requested that the corresponding documentation be removed.

https://github.com/PX4/Devguide/pull/212

JohannesFriis, 2017-07-03:
Hi @priseborough 

I Have been looking i to the fix that was made to resolve the problem and I'm looking forward to flying it during this week.
I have one small remark.

in covariance.cpp  Ekf::resetWindCovariance()

l- 860-861 I would assume that you intend to do a reset to the new uncertainty parameter
	P[22][22] = sq(_params.initial_wind_uncertainty);
	P[23][23] = sq(_params.initial_wind_uncertainty);

I will do some test flights as soon as I figure out how to update a submodule for the Px4 flightstack.

Best Regards
Johannes Friis
jf@sky-watch.com

priseborough, 2017-07-03:
Yes, using that parameter for the reset was intentional.

Something else you might want to try is setting EKF2_MAG_ACCLIM and EKF2_MAG_YAWLIM to 0

That will prevent the switching between mag fusion methods that works well for multi-rotors but had not worked well for some fixed wing logs I have seen recently.

priseborough, 2017-07-18:
Correct operation of sideslip fusion for wind estimation has been verified using replay of http://review.px4.io/plot_app?log=ed369adf-4e46-486a-bc18-14a2ed188bd0

closed 2017-07-18 by priseborough