PX4/PX4-ECL #290: Use baro for ekf alignment

ChristophTobler, 2017-06-29:
I think it would make sense if we stick to baro only for alignment (as we already do for GPS) -> https://github.com/PX4/ecl/blob/7f1cca5b28087055f8343ec069852bedd3b86cd7/EKF/ekf.cpp#L207-L259. Because on many setups the range finder outputs rubbish/measurements outside of max/min when close to ground. Therefore the ekf never aligns.

@priseborough what do you think? Is there something against that?

mhkabir, 2017-06-29:
I ran into this, haha. Took me an hour to figure out why the EKF wouldn't initialise :P

priseborough, 2017-06-30:
@ChristophTobler  Its a  valid option, the baro is the most reliable  sensor on startup. The only other way to handle it I can think of would be to change the range finder driver to return the minimum value if not flying and the reading is outside the valid range.

LorenzMeier, 2017-06-30:
The driver change would mask the missing reading, which would potentially hide real HW issues.

ChristophTobler, 2017-06-30:
@LorenzMeier I agree. And the driver should work independent of the flight status.
So I will change it to use baro if there aren't any objections.

closed 2017-07-03 by ChristophTobler