PX4/PX4-ECL #296: Vision Position constantly re-initializing

lachyd, 2017-07-10:
Using vision position constantly prints out "EKF commencing external vision position fusion", and resets the estimator.

I believe the fix is to add the following directly after line 152 of control.cpp (where it says resetHeight();):

_control_status.flags.ev_pos=true;

This follows the same convention as the ev_yaw conditional below it and works well in simulation.

mhkabir, 2017-07-12:
@lachyd How are you simulating vision estimates in simulation?

lachyd, 2017-07-12:
Using gazebo and ROS. I have a laser scanner mounted on the drone in my gazebo model, which I feed into an SLAM algo and pass the resulting coordinates to PX4 via the mavros vision position topic.

mhkabir, 2017-07-12:
What Z position measurement are you sending with the pose? The scanner only gives you a 2D estimate.

lachyd, 2017-07-12:
I'm using a rangefinder for the Z measurement.

mhkabir, 2017-07-12:
Makes sense. Can you please send in a pull request with your fixes?

lachyd, 2017-07-12:
Done, thanks!

closed 2017-07-13 by mhkabir