PX4/PX4-ECL #299: innovation checks are ignored for get_terrain_vert_pos() return value

CarlOlsson, 2017-07-12:
in ekf2_main.cpp we have
```
lpos.dist_bottom_valid = _ekf.get_terrain_vert_pos(&terrain_vpos);
```

this functions returns true if the terrain estimate is initialized and _range_data_continuous is true.
_range_data_continuous is true if range data is continuously fed into the estimator.

So the lpos.dist_bottom_valid flag is set to true even if the range sensor measurements are failing the innovation checks and are rejected by the ekf.

mhkabir, 2017-07-12:
Was just about to open an issue for this exact thing. Thanks, you beat me to it! 

In order to consolidate terrain following behaviour (i.e controller use the terrain altitude for Z) in range aid mode, we need working correctly. @ChristophTobler.

ChristophTobler, 2017-07-13:
I can have a look at it. Thanks!

ChristophTobler, 2017-07-13:
fixed in https://github.com/PX4/ecl/pull/301

closed 2017-07-13 by ChristophTobler

devbharat, 2017-07-14:
Just as an addendum, note that there is no check to force the measured/estimated height above ground level is positive. 

(This can easily happen if a faulty sensor gives _consistent_ wrong readings (say always close to 3 meters because of fog etc) causing the estimator to reset to the wrong value and propagate from there to values below zero while you are descending)