PX4/PX4-ECL #315: Yaw tracking slowly when using external vision pose

chengshy, 2017-08-01:
Recently, I started playing with PX4 EKF2 for sensor fusion. I really appreciate all the work you have done here!

I tested the combination with magnetometer and optical flow. The state estimation works great in both walking test and flying test. However, I have encountered some problem after I started test External vision pose fusion.

Current setup for me is only mocap pose feed in as external yaw heading and primary height source. 
- EKF2_AID_MASK is set to 16 ( vision yaw fusion)
- EKF2_EV_DELAY is set to 50ms (The data is captured by optitrack system)
- EKF2_DECL_TYPE, EKF2_GPS_CHECK are set to 0. 
- EKF2_MAG_TYPE is set to none (The idea is not using mag at all)
- EKF2_HGT_MODE is set to vision

The height tracking is quite good but yaw track not. When I check the log data, the innovation_check_flag shows a reading 64 very frequently which I believe it means heading rejection. Then I check the innovation of the heading, it seems saturated most time. I tried to set EKF2_EVA_noise as 0.01 at begining because I think mocap system should be fairly accurate. In that  case, the innovation_var for heading is very small. Then I released the EKF2_EVA_noise to 0.3 rad as the default setting for mag_heading as a hack, but the yaw tracking is still not good. I tested both lastest commit at this momment https://github.com/PX4/ecl/commit/c230663b68b8270c15f3e24fd8ca8dc08e7646b1 also this version https://github.com/PX4/ecl/tree/9b6e1ab2a7bf1787842ea9f0acb4008363b5af06.

I am not sure what cause the heading innovation be so big so I am looking for help now. If anyone can give some directions I should look into that will be great.

[image]

[image]

chengshy, 2017-08-08:
@priseborough Can you help me with this. I noticed you are using eular 321 and 312 conversion for quaternion but there is one line seems problematic to me. https://github.com/PX4/ecl/blob/c230663b68b8270c15f3e24fd8ca8dc08e7646b1/EKF/mag_fusion.cpp#L565
When we calculate the yaw 312, should we use Tbn_1_0_neg instead? I am not super sure about this so I want to confirm with you

priseborough, 2017-08-08:
For a yaw, roll, pitch rotation sequence

Tbn(0,1) = -cos(roll)*sin(yaw)
Tbn(1,1) = cos(roll)*cos(yaw)

Therefore

yaw = atan2(-Tbn(0,1),Tbn(1,1))

chengshy, 2017-08-08:
Thanks for your reply. I think my question is not sure whether we should apply transpose here to get the Tbn.

https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/quat2yaw321.m#L11

I am a little confused here about the rotation we are using. Is it extrinsic Z-Y-X rotation for euler321 and extrinsic Z-X-Y for euler312?

priseborough, 2017-08-08:
Yes extrinsic (Tait Bryan)

chengshy, 2017-08-08:
Thanks for that. Is the quaternion_nominal (state 0 -3) is correspond to robot orienation (from body to world)? The quat_to_invrotmat function seems not implementing inverse to me.

priseborough, 2017-08-08:
Yes, from earth to body.

chengshy, 2017-08-09:
Sorry, I am still confused. 
For example, at begining the quaternion is identical [1, 0, 0, 0] (w,x,y,z).  Then we do a yaw rotation pi/2. 
Then the rotation corresponding to quaternion_nominal need to be (from earth to body)
[0 -1 0]          [0  1 0]
[1 0 0]           [-1 0 0]
[0 0 1]    or   [ 0  0 1]  ? 

The corresponding quaternion is [0.707107, 0, 0, 0.707107] and [ 0.707107, 0, 0, -0.707107].

chengshy, 2017-08-09:
And I think function quat_to_invrotmat doesn't calculate the inverse rotation matrix from a quaternion rotation but just the roation matrix corresponds to that quaternion.

priseborough, 2017-11-15:
Please retest using current px4 master and supply a log that starts at boot so we can replicate what your results using the replay function.

closed 2018-08-23 by priseborough