PX4/PX4-ECL #349: Attitude estimate when no gps and magnetic fusion

xmutrue, 2017-10-27:
Will attitude estimate still works fine when no gps fusion and bad magnetic ?

It seems that ECL do not take the accelerometer into observation like ahrs.

Many thanks @priseborough

priseborough, 2017-10-27:
Without any external aiding it falls back to a mode where a static position is fused to constrain drift in roll and pitch. This mode is used before GP checks pass and for flying indoors.

xmutrue, 2017-10-27:
If the fixedwing uav fly outdoors,  and when gps failure, it means that the attitude control of uav can't work well.  
Like other gps/ins product, it will change from gps/ins mode to ahrs mode to provide the pitch and roll for vehicle, and the Controller can use the dead reckoning with airspeed to calculate position.

priseborough, 2017-10-27:
If it has airspeed, and sideslip fusion has also been enabled, then it will dead-reckon. However the PX4 commander failsafe does not yet support this capability fully and will still fall back out of attitude control, requiring the operator to select a non position controlled flight mode. No waypoint controlled flight modes can be used when it is dead-reckoning.

See testing here: https://github.com/PX4/ecl/pull/339

There has been no previous interest in funding development to support use on fixed wing platforms without GPS and airspeed sensing.

closed 2017-10-27 by priseborough

xmutrue, 2017-10-27:
I wonder if the ECL has been port to other platform except the ardupilot and px4.

itayguy, 2023-08-14:
Does PX4 has dead reckoning like ardupilot  for copter ?