PX4/PX4-ECL #357: EKF ringbuffer allocation

dagar, 2017-11-15:
All ringbuffers are allocated in  EstimatorInterface::initialise_interface() https://github.com/PX4/ecl/blob/master/EKF/estimator_interface.cpp#L382-L392

Only allocating each in setups where they're actually used will save a lot of memory.

priseborough, 2017-11-15:
How can I check if a buffer has been allocated?

priseborough, 2017-11-15:
I could add an accessor function to ringBuffer.h that returns false if _buffer == NULL or check that get_length > 0

dagar, 2017-11-15:
Depends how you'd want to do it, but that could work. Based on EKF configuration do we know which are needed or is it dependant on the actual pushed.

An easy option is to allocate on the first push, but that might be questionable if for some reason data only started publishing midflight.

dagar, 2017-11-15:
The ringbuffer is in the class, but the buffer within is allocated with a helper. I could change it so that we only keep pointers to the ringbuffer and the constructor allocates.

Let's figure out how and where the EKF decides it needs a particular ringbuffer and then think about what makes the most sense (including error handling).

priseborough, 2017-11-15:
Another issue is that the ekf2 code assume that all buffers have been allocated and that data (which may be zeros) is always returned. Checking for buffer allocation each time will add to code complexity.

A compromise solution could be to allocate each buffer initially with a length of 1 and then increase the allocation before the first data is pushed to the buffer.

closed 2017-11-17 by LorenzMeier