PX4/PX4-ECL #370: Accel bias is logged as zeros

CarlOlsson, 2017-12-17:
@priseborough 
This log file is from a race quad, is it intended behavior that we set the accel bias states to zero?
[image]
[image]

[logfile](https://logs.px4.io/plot_app?log=b3ba82be-4b0a-46f1-b00b-561abda0536c)
Code was e8624f8afe6199fa8dc9ee1bfc74ac8f6ef52117 with some minor changes to rcS

priseborough, 2017-12-17:
It is expected behaviour to protect against bad learning of bias values if there are high manoeuvre levels. See https://github.com/PX4/ecl/blob/master/EKF/covariance.cpp#L160

closed 2017-12-17 by priseborough

CarlOlsson, 2017-12-17:
Ah okey thanks, I had missed that