PX4/PX4-ECL #376: TECS: Throttle integrator is not initialized at cruise throttle if in air

CarlOlsson, 2018-01-09:
@dagar 
Based on the discussion in https://github.com/PX4/ecl/pull/372 and https://github.com/PX4/Firmware/issues/8508 my understanding is that we want to initialize the throttle integrator with the cruise throttle value when starting TECS when the airplane is already airborne, e.g. at forward transition of a VTOL.

The throttle integrator state is initialized [here](https://github.com/PX4/ecl/blob/master/tecs/tecs.cpp#L534), however, before the value is used it is constrained [here](https://github.com/PX4/ecl/blob/master/tecs/tecs.cpp#L391) leading to a much smaller value

Here is a SITL [logfile](https://logs.px4.io/plot_app?log=6bafd85c-c135-4a3c-bb5a-80fadaa77263) of standard vtol with FW_THR_CRUISE = 0.6
[image]

Is this all good or is it not working as intended?

RomanBapst, 2018-03-19:
@CarlOlsson I don't think we want to initialse the TECS throttle integrator with cruise throttle, see #411 for more information.

Btw, the code link you have shared above refers to airspeed less operation, in which case the throttle integrator is anyway not used.

RomanBapst, 2018-03-19:
Please reopen if you disagree :-)

closed 2018-03-19 by RomanBapst