PX4/PX4-ECL #381: About ekf2 of px4 firmware

ajaybhatt003, 2018-01-23:
Hello Sir,
I want to understand the code of ekf2 of px4 firmware. I am referring to the code provided on site: https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m.
I have two questions regarding the code:

Is it same as ekf2 of px4 firmware?
2)why is time delay horizon technique not used inside matlab code at https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m?
I found time dalay horizon code at https://github.com/PX4/ecl/blob/33e8d5923fc297ef6d7d9ada23cb229bdc1b7836/matlab/EKF_replay/Filter/RunFilter.m
Kindly help me for the same.
Ajay

bkueng, 2018-01-29:
Hi
@priseborough can help.
The ekf2 is documented here: https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html

priseborough, 2018-01-29:
https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m was used to generate the equations in PX4 ekf2.

The delayed time horizon concept is an implementation technique and not part of the EKF state and observation equations.

The time delay horizon method in https://github.com/PX4/ecl/blob/33e8d5923fc297ef6d7d9ada23cb229bdc1b7836/matlab/EKF_replay/Filter/RunFilter.m is only suitable for replay - it uses linear, not circular buffers.

closed 2018-01-29 by priseborough

ajaybhatt003, 2018-01-31:
Hi
@priseborough thank you for reply.
But have implemented delayed time horizon concept in matlab code or simulink then please let me know.
thanks