PX4/PX4-ECL #383: loss of global position - fixed wing flight

Antiheavy, 2018-01-25:
Here is a recent fixed wing flight with a loss of global position at the end of takeoff climb.  Step inputs where throttle down and start to turn; this seems to be a common place where an EKF2 resets occur - at least in our fixed wing application.  Eventually EKF2 sorted itself out and the pilot switched back into Mission mode to complete the test flight without further incident.  I wonder if EKF2 could be made more robust to fixed wing dynamics, or if we have a specific problem with this vehicle.

Here is the log:
https://review.px4.io/plot_app?log=cde7c726-8548-45cf-86cb-fb0f9cac68ba

- EKF2 ended up rejecting the yaw estimate and started ignoring the magnetic compass, it is possible there was a problem there, but we did a fresh compass calibration before the flight and verified north-east-south-west was reasonably close after calibration.  The compass may or may not have been the root-cause.

- The airspeed calibration routine was a little strange prior to this flight.  We did the airspeed calibration 3 times and after each time we would get a preflight error message about the calibration.  Fresh reboot each time and no more warnings. re-cal, then warnings again.  Ultimately the diff pressure measurements on the ground looked correct based on our experience so we flew it.  Airspeed looks reasonable in the log, but there are innovation bit check failures.

Possible other issues seen in this log, but maybe not ECL related:

- Loss of Global Position failsafe triggers immediately instead of waiting a reasonable amount of time for EKF2 to reset.  I think this might be related to eph, epv, and evh thresholds not respecting COM_POS_FS_DELAY.  Maybe this: https://github.com/PX4/Firmware/issues/8260  or maybe something else?

- The vehicle did not seem to switch back to Mission mode automatically once the EKF2 reset.  The pilot switched from Land mode to Mission mode via QGC.  Actually, I'm not sure if the vehicle responded to the pilot's commanded switch or if the firmware finally switched automatically after ~18 seconds.

LorenzMeier, 2018-01-25:
@priseborough Could you have a look?

priseborough, 2018-01-25:
OK will look at it tomorrow morning.

dagar, 2018-01-25:
[cde7c726-8548-45cf-86cb-fb0f9cac68ba.ulg.pdf](https://github.com/PX4/ecl/files/1664358/cde7c726-8548-45cf-86cb-fb0f9cac68ba.ulg.pdf)

[image]

dagar, 2018-01-25:
The vehicle is climbing and at a near maximum commanded bank.

[image]

dagar, 2018-01-25:
The velocity test ratio (dark green) building up during the turn.
[image]

priseborough, 2018-01-25:
I have reviewed the estimator performance and it appears that a build up of yaw error occurred due the transition from heading to 3-axis mag fusion combined with the turn and a combination of errors that exposed a vulnerability that was supposed to have been fixed by https://github.com/PX4/ecl/pull/354

@Antiheavy Can you please provide the git hash for the version of the ecl library used. Based on the the print message just after boot and the presence of the small yaw reset event when switching from yaw to 3-axis fusion, the version you are using appears to predate https://github.com/PX4/ecl/pull/354 which is included in the version used by the PX4 1.7.0 release.

Sequence of events - seconds from boot:

212.5 - Vehicle is launched. 

214 - Vehicle commences a RH bank and achieves a maximum of 29 deg at 217 seconds

215 - EKF switches from heading to 3-axis mag fusion. The yaw change associated with this event is only 2 deg. This type of small yaw reset is unnecessary and due to the temporary reset of quaternion state covariances, makes the EKF attitude estimate vulnerable to measurement errors in the few seconds following. This was fixed by https://github.com/PX4/ecl/pull/354. See plot of quaternion state variances:

[image]

217 - GPS innovations start to rise, positive initially, then large negative. GPs measurements are rejected after 225 seconds. Phase offset between NE components as turn progresses is consistent with a yaw angle error. Note, the GPS velocity innovation test ratio (estimator_status.vel_test_ratio) which takes into account measurement uncertainty and uses all components,  rises progressively from launch.

[image]

235.9 - The EKF times out and resets the velocity, position and yaw using GPS. The yaw angle change due to the reset is -36.9 degrees. After the reset EKF innovations remained low for the remainder of the flight, which indicates that the a buildup of yaw angle error was responsible for the failure of the GPS innovation consistency checks.

The cause of the yaw error is undetermined, but we can rule out major contributors:

Magnetometer calibration. The length of the field vector varies by no more than 5% during the time period from launch to EKF reset once noise spikes are removed. Subsequent variation due to turns is  approx 0.06Ga which is not ideal, but acceptable and typical of PX4 installations (due to lack of soft iron compensation).

Power system interference: Noise spikes of approx 0.03 Ga are present throughout, but while not ideal, this is within the acceptable range.

Bad yaw at launch: The GPS and EKF COG immediately after launch agree and don't start to diverge until after 216 seconds.

[image]

priseborough, 2018-01-25:
@LorenzMeier is there an easy way to record the ecl library hash in logs?

Antiheavy, 2018-01-25:
@priseborough I believe the ECL version is this one which as you suspected predates https://github.com/PX4/ecl/pull/354 by a couple weeks.
https://github.com/PX4/ecl/tree/f921b2de5cbcf309cdf61c2cbe15c0b4dd2ce9b5

Is it possible to EKF replay the data using a more recent ECL version?

priseborough, 2018-01-25:
I will try. Compatibility is more dependant on changes to uORB topic definitions in Firmware.

priseborough, 2018-01-25:
I have replayed using upstream master at commit 757f0e7 and there was no loss of navigation and in-air reset. GPS velocity innovations look good. This confirms the failure hypothesis. 

[image]

Antiheavy, 2018-01-26:
@priseborough good news.  thanks for looking at this.

@dagar any thoughts on the loss of global position failsafe triggering immediately and then not automatically switching back to Mission mode when the EKF reset?  I guess I'm not sure if the switch back to Mission mode after 18 seconds was automatic or a result of the pilot forcing the mode switch...  Does this existing issue cover it  https://github.com/PX4/Firmware/issues/8260 , or should I open a new issue in the Firmware repo?

Antiheavy, 2018-01-27:
I think I can close this issue as current state of ECL does not fail in this case. I will look into another issue on the firmware side if we can figure out exactly why Return mode is triggering immediately instead of giving time for EKF to fix itself.  Thanks all.

closed 2018-01-27 by Antiheavy