PX4/PX4-ECL #384: code writing error in TECS.CPP

yuxiqun, 2018-01-26:
`// Apply limits on the demanded rate of change of speed based on physical performance limits
	// with a 50% margin to allow the total energy controller to correct for errors.
	float velRateMax;
	float velRateMin;

	if ((_uncommanded_descent_recovery) || (_underspeed_detected)) {
		velRateMax = 0.5f * _STE_rate_max / _tas_state;
		velRateMin = 0.5f * _STE_rate_min / _tas_state;

	} else {
		velRateMax = 0.5f * _STE_rate_max / _tas_state;
		velRateMin = 0.5f * _STE_rate_min / _tas_state;
	}`
I think it should not need the "if else",because the one and the other is same.

dagar, 2018-01-28:
Yes, it's useless like this. Would you like to open a pull request to fix it?

LorenzMeier, 2018-01-28:
@priseborough Could you have a look? Should we be doing something else here like e.g. in the if() case not multiply by 0.5?

priseborough, 2018-01-28:
https://github.com/PX4/ecl/pull/386

closed 2018-01-28 by LorenzMeier