PX4/PX4-ECL #400: error in ekf.cpp

asoharev, 2018-02-25:
Hi
I've read the code and it seems like I've found a bug in the velocity calculations at
lines 361-362:
	// compensate for acceleration due to gravity
	_state.vel(2) += _gravity_mss * _imu_sample_delayed.delta_vel_dt;
but _state.vel in Earth frame. I  think you should rotate gravity vector and compensate from full velocity vector.

bresch, 2018-02-26:
Hi @asoharev ,
The rest of the state velocity vector is compensated here https://github.com/PX4/ecl/blob/337cdcc59a2df1f5e395c4c27d69748227e90bbd/EKF/ekf.cpp#L359 using rotated acceleration (https://github.com/PX4/ecl/blob/337cdcc59a2df1f5e395c4c27d69748227e90bbd/EKF/ekf.cpp#L350).
Then the gravity is removed: `_state.vel(2)` is in earth frame, as well as `_gravity_mss` (1g constant). I don't see where the mistake is.

asoharev, 2018-02-26:
I'm sorry, my fault. I understand, EF = NED. 
Before I'd met earth frame as ECEF only (https://en.wikipedia.org/wiki/ECEF) and thought that's generally accepted.

Thanks for your  answer!

closed 2018-02-26 by asoharev

bresch, 2018-02-26:
@asoharev No problem, I agree that it could be quite confusing, I understand why you thought it was wrong then :)