PX4/PX4-ECL #414: Critical loss of GPS with air data aiding issue

priseborough, 2018-03-26:
When GPS is lost and sideslip and airspeed estimation is enabled, the EKF continues to navigate using the wind and sideslip fusion to constrain velocity drift. SITL testing has shown that stability of the air data fusion is inadequate and numerical errors accumulate until air data can no longer be fused. 

[image]

Detected during testing of https://github.com/PX4/Firmware/pull/8771

Preliminary inspection of the SITL log data indicates that this is caused by poor conditioning of the covariance matrix. The following areas need to be reviewed:

1) Wind state process noise
2) Constraint of covariance values
3) Equations and logic for fusion of sideslip and airspeed

priseborough, 2018-04-18:
Fixed by https://github.com/PX4/ecl/pull/413

closed 2018-04-24 by priseborough

thomasgubler, 2018-04-24:
➤ Paul Riseborough commented:

https://github.com/PX4/ecl/pull/413 has been merged