PX4/PX4-ECL #423: Problem with EKF2 and Optical Flow

mubaarik, 2018-04-19:
Hi all,

I am working with Intel Aero ready to fly drone with optical flow and range sensor for an indoor application, no GPS and no barometer. Both the optical flow and the distance sensor data look pretty good with the optical flow showing quality around 150. The distance sensor(teraranger One) is also showing errors of a few centimeters and high sampling rate. The following are the EKF2 parameters I use:

EKF2_ABIAS_INIT 0.20000000298023224
EKF2_ABL_ACCLIM 25.0
EKF2_ABL_GYRLIM 3.0
EKF2_ABL_LIM 0.4000000059604645
EKF2_ABL_TAU 0.5
EKF2_ACC_B_NOISE 0.003000000026077032
EKF2_ACC_NOISE 0.699999988079071
EKF2_AID_MASK 2
EKF2_ANGERR_INIT 0.10000000149011612
EKF2_ARSP_THR 0.0
EKF2_ASPD_MAX 20.0
EKF2_ASP_DELAY 100.0
EKF2_AVEL_DELAY 5.0
EKF2_BARO_DELAY 0.0
EKF2_BARO_GATE 10.0
EKF2_BARO_NOISE 15.0
EKF2_BCOEF_X 25.0
EKF2_BCOEF_Y 25.0
EKF2_BETA_GATE 5.0
EKF2_BETA_NOISE 0.30000001192092896
EKF2_DECL_TYPE 7
EKF2_DRAG_NOISE 2.5
EKF2_EAS_NOISE 1.399999976158142
EKF2_EVA_NOISE 0.05000000074505806
EKF2_EVP_NOISE 0.05000000074505806
EKF2_EV_DELAY 0.0
EKF2_EV_GATE 5.0
EKF2_EV_POS_X 0.0
EKF2_EV_POS_Y 0.0
EKF2_EV_POS_Z 0.0
EKF2_FUSE_BETA 0
EKF2_GBIAS_INIT 0.10000000149011612
EKF2_GPS_CHECK 21
EKF2_GPS_DELAY 110.0
EKF2_GPS_POS_X -0.05999999865889549
EKF2_GPS_POS_Y 0.0
EKF2_GPS_POS_Z -0.10000000149011612
EKF2_GPS_P_GATE 5.0
EKF2_GPS_P_NOISE 0.5
EKF2_GPS_V_GATE 5.0
EKF2_GPS_V_NOISE 0.5
EKF2_GYR_B_NOISE 0.0010000000474974513
EKF2_GYR_NOISE 0.014999999664723873
EKF2_HDG_GATE 2.5999999046325684
EKF2_HEAD_NOISE 0.30000001192092896
EKF2_HGT_MODE 2
EKF2_IMU_POS_X 0.0
EKF2_IMU_POS_Y 0.0
EKF2_IMU_POS_Z 0.0
EKF2_MAGBIAS_ID 396809
EKF2_MAGBIAS_X 0.0
EKF2_MAGBIAS_Y 0.0
EKF2_MAGBIAS_Z 0.0
EKF2_MAGB_K 0.20000000298023224
EKF2_MAGB_VREF 2.499999993688107e-07
EKF2_MAG_ACCLIM 0.5
EKF2_MAG_B_NOISE 9.999999747378752e-05
EKF2_MAG_DECL 0.0
EKF2_MAG_DELAY 0.0
EKF2_MAG_E_NOISE 0.0010000000474974513
EKF2_MAG_GATE 3.0
EKF2_MAG_NOISE 0.05000000074505806
EKF2_MAG_TYPE 0
EKF2_MAG_YAWLIM 0.25
EKF2_MIN_OBS_DT 50
EKF2_MIN_RNG 0.009999999776482582
EKF2_NOAID_NOISE 10.0
EKF2_OF_DELAY 5.0
EKF2_OF_GATE 3.0
EKF2_OF_N_MAX 0.5
EKF2_OF_N_MIN 0.15000000596046448
EKF2_OF_POS_X 0.0
EKF2_OF_POS_Y 0.0
EKF2_OF_POS_Z 0.0
EKF2_OF_QMIN 10
EKF2_OF_RMAX 5.5
EKF2_PCOEF_XN 0.0
EKF2_PCOEF_XP 0.0
EKF2_PCOEF_Y 0.0
EKF2_PCOEF_Z 0.0
EKF2_REQ_EPH 5.0
EKF2_REQ_EPV 8.0
EKF2_REQ_GDOP 2.5
EKF2_REQ_HDRIFT 0.30000001192092896
EKF2_REQ_NSATS 12
EKF2_REQ_SACC 5.0
EKF2_REQ_VDRIFT 0.5
EKF2_RNG_AID 1
EKF2_RNG_A_HMAX 3.0
EKF2_RNG_A_IGATE 1.0
EKF2_RNG_A_VMAX 1.0
EKF2_RNG_DELAY 20.0
EKF2_RNG_GATE 5.0
EKF2_RNG_NOISE 0.30000001192092896
EKF2_RNG_PITCH 0.0
EKF2_RNG_POS_X 0.0
EKF2_RNG_POS_Y 0.0
EKF2_RNG_POS_Z 0.0
EKF2_RNG_SFE 0.05000000074505806
EKF2_TAS_GATE 3.0
EKF2_TAU_POS 0.25
EKF2_TAU_VEL 0.25
EKF2_TERR_GRAD 0.0
EKF2_TERR_NOISE 0.5
EKF2_WIND_NOISE 0.10000000149011612

Below is a screenshot of the mavlink inspector as well:

[image]

The problem is that while the optical-Flow and the range data is clearly published, the EKF is not using the optical-flow and the distance sensor for state estimation. The analysis of collected logs show that the EKF is using the Barometer for height estimation (although we set the range sensor as primary aid) and that the optical flow innovations stays around zero. 

Any suggestions?

Thanks a lot!

Abhi9499, 2018-08-06:
Even I am getting same issue. Everything seems to be fine and data is flowing but EKF2 is not using that. There is no sign of Flow init message. Fortunately I got everything running last time may be 3 to 5 months ago.

priseborough, 2018-08-23:
Please post user support issues and logs here: http://discuss.px4.io/c/support

I recommend you log from boot and learn how to use the ekf2 replay function https://dev.px4.io/en/debug/system_wide_replay.html#ekf2-replay to do diagnostics on your setup.

closed 2018-08-23 by priseborough