PX4/PX4-ECL #427: EKF state time offset (potential bug?)

Changliu52, 2018-04-25:
Hi @priseborough ,

I understand EKF2 runs on the delayed tine horizon, and the delay buffer is set according to the highest delay of the sensors. I assume the state of EKF will fuse the sensor according to their timestamps.

So I set ASP_DELAY to 250ms. I then run the vision slam with hardware sync against px4 camera trigger.  However, in my log from flightPlot, although the timing from local_position matches the timing from vision_position (graph overlaps pretty well), the ekf_state[7] and ekf_state[8] plot is delay shifted about 250ms in the plot. 
[image]

I assume the flightPlot plots the graphs according to their timestamps. If that is the case, does that mean the timestamp of the ekf_state is wrong? and whether the updates were fused with the corrected relative timestamps?

Also I noticed that pos_test_ratio is goes up in during motion, although the innovation seems quite flat during motion.

Thank you in advance.
Chang

priseborough, 2018-04-25:
The timestamp on the estimator_status log message that contains the states is the time the message was logged, not the time of the EKF's internal time horizon. For tuning you adjust the vision delay parameter or adjust the time stamp offset so that the change in innovation with velocity changes is minimised.

closed 2018-04-25 by priseborough

Changliu52, 2018-04-26:
I see. Thanks Paul.

Chang