PX4/PX4-ECL #446: catkin build fails - header files not found

nolanholden, 2018-05-11:
Upon building with catkin_tools (`catkin build`), px4 cannot find the header files, suggesting that the CMake config is somehow not communicating the include path of `px4-firmware/src/lib/ecl/`

Any ideas on how to change the cmake files?

I am on latest from master.

*Note*: I have forked this repository and renames the root dir as `px4-firmware` from `Firmware`.

```
Errors     << px4:make /home/me/dev/mavros-ws/logs/px4/build.make.001.log              
In file included from /home/me/dev/mavros-ws/src/px4-firmware/src/modules/fw_pos_control_l1/FixedwingPositionControl.hpp:58:0,
                 from /home/me/dev/mavros-ws/src/px4-firmware/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp:34:
/home/me/dev/mavros-ws/src/px4-firmware/src/lib/ecl/l1/ecl_l1_pos_controller.h:65:21: fatal error: geo/geo.h: No such file or directory
compilation terminated.
make[2]: *** [src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/FixedwingPositionControl.cpp.o] Error 1
make[1]: *** [src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/me/dev/mavros-ws/build/px4; catkin build --get-env px4 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

dagar, 2018-05-22:
I believe this has been resolved. Can you try again?

closed 2018-05-22 by dagar

ana-GT, 2018-05-28:
Hi. I am experiencing the same issue (build fails due to not finding ecl-related header files). This issue happens when using "catkin build" to build the px4 as a ROS package (if I just build it as a regular CMake project it builds fine, so I suspect the error comes from some catkin_tools magic):

/home/ana/ws/src/px4/src/lib/ecl/attitude_fw/ecl_controller.h:51:17: fatal error: ecl.h: No such file or directory

dagar, 2018-05-28:
@nolanholden @ana-GT @TSC21 please open an issue in PX4 Firmware and describe exactly how you want to use this. We need to add a basic catkin build test to the continuous integration system (http://ci.px4.io:8080/blue/organizations/jenkins/pipelines) if we want to keep things like this from breaking all the time.

nolanholden, 2018-05-29:
@ana-GT I found no solution but realized I did not need to build it using catkin_tool

Just use the Make targets from px4. (for me it was `make posix_sitl_default gazebo`). [Instruction at my repo here for anyone in need.](https://github.com/redbirdrobotics/redbird_quadrotor#build-the-gazebo-simulation-and-mavros-node)

ana-GT, 2018-05-29:
Hi Nolan, well, this is by no means fixed, but my workaround was: Revert to the last stable release for px4 (which calls in turn a ecl's specific commit which builds fine). 

I know that I could just use the plain make and do all that env stuff, but I prefer to only interact with catkin_tools and not to mess directly with environment variables if possible (in my book that is a bit messy, but that is a personal preference). 

I managed to have px4 + sitl_gazebo installed fine as a ros packages (I had to tweak a tiny bit the sitl_gazebo CMake to allow it to find Ogre when being built with catkin_build, but other than that, the process was smooth).

pratyusv, 2018-06-06:
@dagar I am also facing the same issue. I have cloned the latest px4 repo.

pratyusv, 2018-06-14:
@mickey13 Thanks. I did the same. It was working. Do you have a rough idea why it's failing?

dagar, 2018-06-15:
There's a problem with the include path. For some reason this is only working outside of catkin. https://github.com/PX4/ecl/blob/master/attitude_fw/CMakeLists.txt#L43 

Keep an eye on https://github.com/PX4/Firmware/pull/9679 for a solution.