PX4/PX4-ECL #459: Timestamp comparison issue

bkueng, 2018-06-13:
[here](https://github.com/PX4/ecl/blob/master/EKF/control.cpp#L615) we have the check:
```
_time_last_imu - _time_last_gps > (uint64_t)10e6
```
And from `ekf2_main` the gps data is copied and set via `setGpsData`, after `setIMUData` is called. It means that `_time_last_gps > _time_last_imu` can happen. And the comparison is unsigned, so the check will result true.

This happens in SITL for example, the timestamp difference is only a few microseconds.
I see more of these checks, so the problem exists elsewhere too.

Possible solutions:
- convert all checks to use signed logic
- convert all checks like this: `_time_last_imu > _time_last_gps + (uint64_t)10e6`
- call `setIMUData` after all other data is already set. This might have other side-effects
- make sure `_time_last_imu` is always the last timestamp. We would then set the gps timestamp like this:
  ```
  _time_last_gps = min(_time_last_imu, time_usec)
  ```

The last one is the simplest solution with probably the least side-effects.

@priseborough what do you think?

mhkabir, 2018-06-13:
Also this was merged without much discussion https://github.com/PX4/ecl/pull/450, but I don't have clarity on what it fixes/attempts to fix. Looks like a similar issue.

LorenzMeier, 2018-06-14:
The last one looks good.

priseborough, 2018-06-14:
This check requires GPS use to be true which means previous use of GPS. This can only happen if GPS checks have been passed which requires 10 seconds of valid GPS data retrieved from the buffer.  Also data is only retrieved from the buffer after it has fallen behind the fusion time horizon which is delayed relative to _time_last_imu by the length of the IMU buffer.

priseborough, 2018-06-14:
There are three timestamp error conditions that will affect the operation of the EKF:

1) The data timestamp is greater than than _imu_sample_delayed.time_us + 1000 * FILTER_UPDATE_PERIOD_MS. If this happens data will be fused at the wrong time because the fusion time horizon will have already passed before the EKF has opportunity to retrieve the data from the buffer.

2) Data is out of order, This will cause data to be fused at the incorrect time because the buffers require the timestamps to be always increasing.

3) The time stamp is small enough or the arrival data rate is high enough such that the data has been overwritten with new data before it can be retrieved from the buffer.

priseborough, 2018-06-18:
Fixed by https://github.com/PX4/ecl/pull/462

Will keep this open until the other instances, eg range finder have been fixed.

priseborough, 2019-04-04:
@bkueng has the introduction of lock-step scheduling fixed the likelihood of this happening with other sensors during SITL use?

jkflying, 2019-04-04:
@julianoes might be a better person to ask, I think he did most of it

julianoes, 2019-04-04:
With lockstep the timestamps are now regular every 4000 us so this case should not be triggered anymore except in the very beginning when the systemk "freewheels" until initialized. We might make the locksteping granularity finer (e.g. 1000 us) but still that should not trigger issues.

However, even with lockstep, a comparison like this should potentially be hardened for this case.

bkueng, 2019-04-04:
Exactly, lockstep makes that now harder to trigger and detect.

PR #729 (2020-01-28): Robustify timestamp checks
This PR solves the open issue on timestamp checks: fixes #459
The issue was previously addressed by lower-bounding the timestamp of incoming GPS and baro samples. This can potentially be dangerous as rather old data could be re-timestamped and still be used. By removing these lower bounds old data would enter the buffer as it is and would be rejected by the Ringbuffer [checks](https://github.com/PX4/ecl/blob/master/EKF/RingBuffer.h#L133).

The issue of comparing unsigned integers is solved by avoiding the subraction. e.g:
```
_time_last_imu > _time_last_gps + (uint64 [... cut]
closed 2020-01-28 by priseborough