PX4/PX4-ECL #478: EKF continues to use last range data

priseborough, 2018-07-12:
If range data stops publishing, the EKF continues to use the last data when EKF2_RNG_AID is enabled

https://review.px4.io/plot_app?log=861edb66-663c-46b2-8a60-d1935a7290b6

https://review.px4.io/plot_app?log=89526d09-77bf-4edf-9a02-7c7c34e62681

priseborough, 2018-07-12:
The practice of indicating invalid or out of range data by not publishing as seen in these logs is not recommended. The code will be updated so that loss of range data causes reversion to use of the default height sensor, but because of the delay associated with a timeout check this mechanism should not be relied on for normal switching.

dakejahl, 2018-07-12:
Thanks, we will merge #9865 on our side to use the `signal_quality` field.

priseborough, 2018-07-13:
Prototype fix here: https://github.com/PX4/ecl/tree/pr-ekfRngAidFix

Will raise PR when testing complete.

dakejahl, 2018-07-13:
Hey Paul, I merged in this commit and went testing **(Note:** I did not change the way our driver is publishing for the sake of simply testing the timeout on the fusing of the data). 
https://review.px4.io/plot_app?log=46f674ad-abbc-41a3-8e62-2a0602417479

I noticed you have a variable `RNG_MAX_INTERVAL` which is set at 200ms that is used to determine if the data has gone stale. 

```
	if ((_imu_sample_delayed.time_us - _range_sample_delayed .time_us) > 2 * RNG_MAX_INTERVAL) {
		_rng_hgt_faulty = true;
		return;
	}
```
After 400ms we should realize that we have stale data. Here are some data snapshots from this log:|
[image]
**Timestamps for the 3 data points**
[image]
[image]

Notice the range_aid does not actually get disabled until it gets the data point at 4meters, which is a time delta of 2.7 seconds, much higher than the expected 400ms. Thanks for the help with this!

dakejahl, 2018-07-13:
Please see [#480](https://github.com/PX4/ecl/pull/480) 

Thanks!