PX4/PX4-ECL #479: EKF2 does not detect if no mag is fused

ndepal, 2018-07-13:
We had an incident where the mag data published in `sensor_combined` was not updating properly. The timestamps `sensor_combined_0.magnetometer_timestamp_relative` showed that the data was old and EKF2 did not do any fusion.

However, it still claimed that there are no faults / that the position solution was good.
Position controlled flight performed poorly (significant toilet-bowling).

In my opinion, EKF2 should mark the position solution as bad if no heading fusion was done within some timeout.

Here is a log file: https://logs.px4.io/plot_app?log=8f1505e6-f998-47e4-9693-58c3c0424774

@dagar as per our discussion.
@priseborough what do you think?

dagar, 2018-07-13:
This is firmware side, but I'm also wondering about tightening the commander check.
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1794-L1820

dagar, 2018-07-13:
[8f1505e6-f998-47e4-9693-58c3c0424774.ulg.pdf](https://github.com/PX4/ecl/files/2193533/8f1505e6-f998-47e4-9693-58c3c0424774.ulg.pdf)

ndepal, 2018-07-17:
I can re-create the situation in SITL on current master with the following change in PX4/Firmware:
https://github.com/ndepal/Firmware/commit/172815f621be649d094fdc38fb4aaef978eb4577

This will cause EKF2 to stop fusing mag measurements after 2000 published mag measurements. This causes heading drift in simulation and subsequent control instability* (having done `commander takeoff` before the above triggered).

PX4 remains in position hold mode (and allows taking off) in this situation, despite having detected a mag timeout:
```WARN  [sensors] Cnt == 2000
ERROR [sensors] Mag #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Mag #0 priority: 1
```

I have tried to make EKF2 detect the heading fusion timeout with the following change in PX4/ecl:
https://github.com/ndepal/ecl/commit/a15bce935167b0ba3a8a4967bc12d6f124d1293a

EKF2 does detect the timeout, sets `_fault_status.flags.bad_mag_hdg` to true, and `estimator_status.solution_status_flags` goes to 192 (i.e. neither `True if the horizontal position (relative) estimate is good` nor `True if the horizontal position (absolute) estimate is good`).

However, PX4 still does not go into a failsafe mode.

\* Since the heading drift depends on the random gyro noise/drift, the timing and magnitude of the control instability varies between simulation runs.

ndepal, 2018-07-23:
@priseborough I would really appreciate your input on this.

priseborough, 2018-07-23:
I will have a look at it this week

priseborough, 2018-07-23:
In the scenario posted, the velocity and position errors are not large enough to fail innovation consistency checks. position and velocity errors are therefore within limits and the ekf has no reason to declare the local_position data invalid.

[image]

The toilet bowling is a predominantly due to a coupling between estimator yaw error that cause transient velocity errors and the velocity controller responses in body frame. Normally when yaw errors build up to the point where toilet bowling starts, the motion makes the yaw observable and the motion subsides or stabilises.

I do not believe this behaviour is something that we should handle with addtional EKF checks that declare local_position invalid because the sensitivity to yaw errors is vehicle type specific. Additional vehicle and controller type specific checking should be done in the commander.

ndepal, 2018-07-24:
Thanks for having a look at this, Paul.

> Normally when yaw errors build up to the point where toilet bowling starts, the motion makes the yaw observable and the motion subsides or stabilises.

This does not match my SITL tests described above, where the vehicle reproducibly crashes into the ground after some time.

If I understand correctly, you are suggesting that mag failures should be detected by the increase in position/velocity innovations caused by the heading estimation error.
I find this problematic because:
1. It requires that the vehicle does some toilet-bowling to make the heading error observable. Depending on the environment this can be dangerous. In my SITL tests, the vehicle did not stop circling until it (inadvertently) hit the ground.
2. Tightening the position innovation gate may have other negative side effects, especially in environments with potentially jumpy GPS measurements, where a wider gate may be necessary.
3. It does not allow detecting mag failures on the ground before flight.

priseborough, 2018-07-24:
During work on a recent commercial multi-rotor project I was involved with, the behaviour of the EKF and position controller for varying levels of takeoff yaw error was extensively flight tested and in that application, the yaw and toilet bowling was found to be self correcting.

I was recommending that declaring local_position invalid on the basis of mag fusion timeout  should not be done by the EKF. For example a FW plane or copter doing regular speed and/or course changes can fly without mag fusion.

I do not have a problem with strengthening the preflight checks in the commander to include a check for mag timeout. Adding it to in-flight failsafes does need to be application/vehicle type specific. Making a FW vehicle failsafe due to a mag timeout is more risky than allowing it to continue.

ndepal, 2018-07-24:
> During work on a recent commercial multi-rotor project I was involved with, the behaviour of the EKF and position controller for varying levels of takeoff yaw error was extensively flight tested and in that application, the yaw and toilet bowling was found to be self correcting.

Unfortunately I have not had the same experience.

> For example a FW plane or copter doing regular speed and/or course changes can fly without mag fusion.

Understood. So you're proposing that the preflight check applies to all vehicles and in-flight failures depend on vehicle type or some parameter?

I think a fundamental problem is that the "failover" that happens when only one mag is present effectively does nothing at all because it fails over to the sensor that just failed.

Perhaps the behavior in general when there are no more sensors to fail over to needs to be defined. Just like we can configure e.g. no action on telemetry loss, we could add an option on what to do on total mag failure.

closed 2018-12-20 by priseborough