PX4/PX4-ECL #492: Range aid: use tilt angle not horizontal velocity

dakejahl, 2018-08-08:
@priseborough I'd like to change **range aiding**s dependence from `horizontal_velocity` to `tilt_angle`. I think it makes the most sense for each range sensor to publish a `max_tilt` in the `distance_sensor_msg` because some range sensors need to be normal to the surface (laser), while others are conical (sonar) and do not need to be. The dependence on `horizontal_velocity` seems odd, because what about a FW vehicle that is moving very fast but would still like to make use of a range sensor to support the baro? 

I wanted to ask you first to get your opinion/advice before doing the work and opening a PR.

Thanks!
Jake

dakejahl, 2018-08-08:
On a similiar note, there are a few other things I'd like to tweak about the way we do range aiding.

1. Allow range aiding while `landed` as long as the measurements are within `min_distance` +/- `some_delta`. This means the distance sensor drivers supply a minimum distance that is also the `height_to_ground_offset` applied against the measurements. I'll explain why in (2).
2. If we have good range finder readings at `takeoff` then run a complementary filter over the `baro_hgt_offset` using the rangefinder data as the reference for the baro datum. This is important because the baro offset is _usually_ wrong (I've got data to prove this). Especially when you are dealing with things like cool tall grass vs hot pavement (air density variations are large). 
3. Remove arbitrary and generalized limitations and instead require the driver to supply (or not supply) these values. i.e. tilt angle, max velocity, etc

dakejahl, 2018-08-09:
I have a few more ideas:
1. When not using GPS as primary, periodically update GPS `alt_ref` to bring it into alignment with the current state prediction (taking alt ref on ground is pretty unreliable due to gps multipath)
2. Update `_baro_hgt_offset` to bring into alignment with current state prediction when switching from **range aiding** back to baro. 
3. In order to accomplish (1) and (2) we need to be tracking in a global variable a moving average of the measurements for baro, gps and range finder and use that as the measurement we compare against the state to set a new offset. 
4. Rather than seeding the `ekf` `local_z_pos` with sensor measurements while `landed` we should be seeding it with `0` and then calculating our sensor offsets from there. I've seen in both our flight logs and others the `local_z_pos` starting at some nonsensical value such as -4 at takeoff. 

Sorry for bombarding you this week, I wanted to share with you what I've discovered this week. I feel very strongly about the importance of good datums and agreement between sensors, as this ultimately effects the quality of our estimate and the consistency of flight performance.

priseborough, 2018-08-11:
Ultimately, if we want to fude data from disparate vertical position sensing mechanisms I think we need a  separate estimator to estimate the height datum offsets that incorporates knowledge of the different error mechanisms for each sensor.

dakejahl, 2018-08-11:
I agree! I've got a few other things slated for the next few weeks, but I'd like to tackle that afterwards. Is this something you'd condone? I can do all the work but I'd like to solicit your advice during the process. Timeline is probably 6 weeks from now and I'd dedicate 2-3 weeks to the rework.

priseborough, 2018-08-23:
OK, we have more than one issue being discussed here so I will summarise my responses:

1) I do not support removing dependence of range aid on velocity due to terrain sensitivity issue raised above.
2) I do support addition of a separate estimator to enable blending of various height sources.

If you wish to undertake 2) please raise a new issue for it (enhancement), keep the changes as modular as possible, treating the blended product as a separate height measurement type so that the changes to exisiting logic and fusion methods are minimised.

closed 2018-08-23 by priseborough