PX4/PX4-ECL #511: ekf2 local position estimates vz down when maintaining altitude

kyuhyong, 2018-09-13:
I thought this problem has gone after replacing the fc board and power module however its keeps happening. 

In altitude hold mode, drone should maintain its altitude when throttle is about 50%. 
But in the log below, at about 1:25 seconds in to the flight, drone think its z velocity going down (+z in NED frame) while sonar measurement shows no indication of going down. 
Sonar is enabled but not did not set ekf2 range aid mode. 

[image]

https://logs.px4.io/plot_app?log=bbf99ec1-a980-4338-b9ca-1b7cb18db286

Where should I look at to solve this problem?

RomanBapst, 2018-09-13:
@kyuhyong Some observations that might help you:
1) Yes, there is a large inconsistency between the height state and the height rate state. Also the barometer innovation is large and biased towards the end of the log. You need to find out if the problem is the baro or if the inconsistency comes from the IMU (e.g. clipping). If possible you can switch to using range finder altitude or gps altitude.

2) You have large heading innovations. Your magnetic field factor is not constant in magnitude at all.
If you compass calibration is done properly and you don't get interference from the drones power system then it's probably interference from the environment.

kyuhyong, 2018-09-13:
> @kyuhyong Some observations that might help you:
>
> 1. Yes, there is a large inconsistency between the height state and the height rate state. Also the barometer innovation is large and biased towards [...]

Hello RomanBapst, 
I appreciate your response. 
1. For the above flight, I used range finder measurement as height reference for altitude hold while vz for velocity control (Up and down in alt hold mode). GPS altitude is not available since the drone is for indoor. 
If this problem is due from barometer, what can be done to solve? I already placed a thin sponge between sensor and fc cover and when looking at the barometer altitude, it looks not bad.

2. In the lab I used to fly the drone, floors are covered with square panels which I think is made of steel thus making a lot of interference. Does magnetometer affects height estimation too? 
One more question. 
Is ekf2 estimator can run without magnetometer at all and only using gyro? 
Thank you,

Kyu

kyuhyong, 2018-09-20:
It turned out that putting more sponge tape around fc actually improves the vz estimation as shown here.
https://logs.px4.io/plot_app?log=e94777f3-119d-4218-9fba-3d529b0d2f23
Thanks for looking at the log. ;)

kyuhyong, 2018-09-24:
I'm still have this issue. 
https://logs.px4.io/plot_app?log=c96d25ad-e2f3-422d-851e-b294ab397ff5
It seems like once vz estimation diverge, it takes too long to get back.

huangwen0907, 2018-09-25:
BTW:
It seems that your controllers is not response to your commander: May you check your hardware, such as esc or motors. etc

[image]

Why set `EKF2_AID_MASK  2` to inhibit the imu bias learning.

kyuhyong, 2018-09-26:
That happens when vz estimation is inconsistent with z estimation. ie vz says the drone is going down when absolute altitude of drone is going up. 
To my understanding, setting 2(bit 1) is for using optical flow and inhibit IMU is for setting 4 or bit 2.
[image]

closed 2018-12-20 by priseborough